dji-sdk/Onboard-SDK #357: Receiving Data sent from MSDK on the OSDK

LarryYourWaiter, 2018-07-31:
Hello all,

I need to receive a message sent from the MSDK, on the OSDK. I looked at the [mobile communication API reference](https://developer.dji.com/onboard-api-reference/classDJI_1_1OSDK_1_1MobileCommunication.html), and at the mobile communication samples. But I was still confused on how to make this work. As of now, I have a function called 

**getDataFromMSDKCallback(Vehicle(there is an asterisk here) vehiclePtr, RecvContainer recvFrame, UserData userData)**

 that translates **userData** from a byte array to individual characters (then those characters are used later on). My issue is that this function is not being called. Is there an example of how to receive data from the MSDK (not commands)? And how do I get the OSDK to constantly listen for incoming MSDK messages?

Thank you.

benUrsel, 2018-08-02:
Have you been able to send a message from the MSDK to the OSDK? I have been able to send from MSDK to OSDK, but not the other way around.

imkhairulikhwan, 2018-08-03:
@benUrsel I had the similar issue before, did you refer to:

https://developer.dji.com/onboard-sdk/documentation/sample-doc/msdk-comm.html

to be able to send from MSDK to OSDK?

Thanks.

jonathanmichel, 2018-08-03:
You have to set a callback function that will be called on data reception from MSDK. Here is an example with a class implementation:

Mobile.h
```
class Mobile {
    public:
       explicit Mobile();

        /**
         * Define callback function for parsing incoming data
         */
        void setup();

        /**
         * Callback for incoming mobile data. Process data and
         * decide what to do with them
         * @param vehicle Vehicle receiving data
         * @param recvFrame Incoming data
         * @param userData Mobile object casted as void*
         */
        static void mobileCallback(Vehicle *vehicle,
                                   RecvContainer recvFrame,
                                   UserData userData);
};
```

Mobile.cpp
```
void Mobile::setup() {
    // Mobile object is passed as parameter to grant access to object attributes
    DSTATUS("MOBSDK Setup");
    vehicle->moc->setFromMSDKCallback(mobileCallback, (void *) this);
}

void Mobile::mobileCallback(Vehicle *vehicle, RecvContainer recvFrame,
                            UserData userData) {
    // Cast userData to Mobile object to have access to object attributes
    auto *m = static_cast<Mobile *>(userData);

    uint8_t formatFrameLength = OpenProtocol::PackageMin + OpenProtocol::CRCHead;
    uint16_t msgLength = recvFrame.recvInfo.len - formatFrameLength;
    uint8_t* data = recvFrame.recvData.raw_ack_array;
 
        // Example : Data received displayed as string
        if(msgLength == 100) { 
            msgLength = 99;
        }
        data[msgLength] = '\0';
        DSTATUS("String received : %s", recvFrame.recvData.raw_ack_array);

    // You can process your received data here
}
```
All you have to do is to call setup() method from main.

benUrsel, 2018-08-03:
@imkhairulikhwan Yes I was able to get that sample to work properly, but am having the issue when I implement it myself in swift. I believe it's a problem with how the app handles the call to

<code> func onboardSDKDevice(_ osdkDevice: DJIOnboardSDKDevice, didSendDataToMobile data: Data)
</code> 

I can send data from mobile, receive data onboard and send data from onboard. The crash happens when the mobile receives data.

ViewController.h

<code>

    import UIKit
    import DJISDK

    class ViewController: UIViewController, DJISDKManagerDelegate, DJIOnboardSDKDeviceDelegate {
    var onboardSDKDevice: DJIOnboardSDKDevice?
    
    @IBOutlet weak var labelView: UILabel!
    @IBOutlet weak var dataDisplay: UILabel!
    
    @IBAction func ConnectToDrone(_ sender: Any) {
        
        //creates 4 bytes of zeros
        let var1 = Data(count: 4)
        onboardSDKDevice?.sendDataFromMobile(toOnboard: var1, withCompletion: nil)
    }
   
    override func viewDidAppear(_ animated: Bool) {
        
        super.viewDidDisappear(animated)
        DJISDKManager.registerApp(with: self)
    }
    
    override func viewDidLoad() {
        super.viewDidLoad()
    }
    
    override func didReceiveMemoryWarning() {
        super.didReceiveMemoryWarning()
    }
    
    func showAlertViewWithTitle(title: String, withMessage message: String) {
        let alert = UIAlertController.init(title: title, message: message, preferredStyle: UIAlertControllerStyle.alert)
        let okAction = UIAlertAction.init(title:"OK", style: UIAlertActionStyle.default, handler: nil)
        
        alert.addAction(okAction)
        self.present(alert, animated: true, completion: nil)
    }
    
    // DJISDKManagerDelegate Methods
    func productConnected(_ product: DJIBaseProduct?) {
        //when product connects set the label and create and onboard object
        NSLog("Product Connected")
        labelView.text = DJISDKManager.product()?.model
        onboardSDKDevice = DJIOnboardSDKDevice()
        onboardSDKDevice?.delegate = self
        if(onboardSDKDevice != nil){
            dataDisplay.text = "Delegate set"
        }
    }
    
    func productDisconnected() {
        NSLog("Product Disconnected")
        labelView.text = "Product disconnected"
    }
    
    func appRegisteredWithError(_ error: Error?) {
        var message = "Register App Successed!"
        
        if (error != nil) {
            message = "Register app failed! Please enter your app key and check the network."
        }else{
            NSLog("Register App Successed!");
        }
        
        self.showAlertViewWithTitle(title:"Register App", withMessage: message)
    }
    
    //DJIOnboardSDKDevicDelegate Methods
    
    //This is for recieving data sent from the drone
    //This crashes app every time it is called
    func onboardSDKDevice(_ osdkDevice: DJIOnboardSDKDevice, didSendDataToMobile data: Data) 
    {
        dataDisplay.text = "recieved data"
        }
    }
</code>

LarryYourWaiter, 2018-08-06:
Thanks for the responses, I happened to solve it before I saw these. I just forgot to call the setupMSDKParsing function. If anyone has the same issue I am more than happy to help.

jonathanmichel, 2018-08-06:
@LarryYourWaiter You're welcome

@benUrsel Yes I receive data on my Android app. I use Linux and I had just to call `vehicle->moc->sendDataToMSDK(data, length);` on Onboard computer and implement the `setOnboardSDKDeviceDataCallback` function in my Android application, as explained [here](https://developer.dji.com/onboard-sdk/documentation/appendix/mobile-onboard-APIs.html). But it seems you're using IOs, I'm affraid I can't help you :(

jonathanmichel, 2018-08-07:
@imkhairulikhwan Are you talking about[ Linux Onboard Mobile sample](https://github.com/dji-sdk/Onboard-SDK/tree/3.6/sample/linux/mobile) ? If yes, it seems that it is designed to work with the [iOS sample application](https://developer.dji.com/onboard-sdk/documentation/sample-doc/msdk-comm.html). I did not try it.

I'm not totally sure to understand which system you are using on both side (Onboard and Mobile) and what you attempted to do ?

If you want to receive data on a Linux OnBoard computer, take a look at my first reply.

If you want to send data from a Linux Onboard computer, you are supposed to have a Vehicle object. You can simply call `vehicle->moc->sendDataToMSDK(reinterpret_cast<uint8_t*>("some data"), 9);` (9 is data length) to send data to Mobile.

If you want to receive data on an Android app here is a code that works :
```
// registrationCallback is an attribute of the main activity class.
private DJISDKManager.SDKManagerCallback registrationCallback = new DJISDKManager.SDKManagerCallback() {
        @Override
        public void onRegister(DJIError error) {
            isRegistrationInProgress.set(false);
            if (error == DJISDKError.REGISTRATION_SUCCESS) {
                DJISDKManager.getInstance().startConnectionToProduct();
                toast("SDK registration succeeded!");
                loginAccount();
            } else {
                toast("SDK registration failed, check network and retry!");
            }
        }

        @Override
        public void onProductChange(BaseProduct oldProduct, BaseProduct newProduct) {
            if(newProduct != null) {
                toast("Aircraft connected");
                Aircraft mAircraft = (Aircraft)newProduct;
                mFlightController = mAircraft.getFlightController();
                if(mFlightController != null) {
                    mFlightController.setOnboardSDKDeviceDataCallback(new FlightController.OnboardSDKDeviceDataCallback() {
                        @Override
                        public void onReceive(byte[] bytes) {
                            StringBuilder buffer = new StringBuilder();
                            for (byte b : bytes) {
                                buffer.append((char) b);
                            }
                            toast(buffer.toString());
                        }
                    });
                }
            } else {
                toast("Aircraft disconnected");
            }
        }
    };
```

And finally, to send data from Android, you can use this function (implement it in your main activity).

```
public void sendData(byte[] data) {
    if(mFlightController != null) { 
        mFlightController.sendDataToOnboardSDKDevice(data, new CommonCallbacks.CompletionCallback() {
                @Override
                public void onResult(DJIError djiError) {
                    if(djiError != null) {
                        toast(djiError.toString());
                });
}
```

Toast method is just a debug method that display a toast message and print it in the console of Android Studio. Note that sometimes toast messages doesn't display on Android, you have to [enable notification permission](https://www.google.ch/search?q=enable+app+notifications+android&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjmpYaY5drcAhXQjqQKHbAkBi8Q_AUICigB&biw=1920&bih=983#imgrc=5NSeKUefGMQkpM:) for your app.
```
    public void toast(final String text)
    {
        ((Activity)this).runOnUiThread(new Runnable() {
            @Override
            public void run() {
                System.out.println(text);
                Toast.makeText(getApplicationContext(), text, Toast.LENGTH_SHORT).show();
            }
        });
    }
```

If you are using iOS I can't help you.
If you are using an other Onboard computer platform it is not supposed to be (too) different (at least I think).

Hope it helps

imkhairulikhwan, 2018-08-08:
@jonathanmichel Thanks for the detailed description! really appreciate it. Yes, I was referring to the provided OnBoard Mobile sample. Should have known it is designed for iOS :/ 

I am using Jetson TX2 for OnBoard and attempt to initiate communication from both sides (mobile and OnBoard) so that we could have better control on the drone. 

I will try the samples and update the result here. Thank you again!

imkhairulikhwan, 2018-08-08:
@jonathanmichel Also, may I ask if you have tried the waypoint missions? 

I was wondering if we could use the waypoint missions to execute autonomous flight. If so, would this sequence be correct? Once finished each mission, we programmatically add a new mission (with multiple waypoints) and run it again?  

The documentation seems available here http://wiki.ros.org/dji_sdk but the links provided inside the website are not available (broken links). 

Thank you!

jonathanmichel, 2018-08-08:
@imkhairulikhwan You're welcome. Good luck, it should work if you use the code above.

Concerning the waypoint mission, yes you can add GPS coordinates (long, latt, alt) from the Onboard computer and play a sequence. You can then pause, resume and stop the mission. I'm currently working on this with my project. Take a look at [this Linux sample ](https://github.com/dji-sdk/Onboard-SDK/tree/3.6/sample/linux/missions). The thing I didn't found yet is how to get the state of the mission to know when a point is reached or the mission done. I you find something, I'm curious.

It seems that you can launch a waypoint mission from the Mobile SDK too, according to [this tutorial](https://developer.dji.com/mobile-sdk/documentation/android-tutorials/GSDemo-Gaode-Map.html). I didn't try this sample yet.

imkhairulikhwan, 2018-08-08:
@jonathanmichel we actually implemented waypoint mission from the Mobile SDK before, but due to some requirements, we need to integrate it with Onboard SDK as well. Hence, Mobile SDK is planned to act only to send the command instead of initiating the missions.

souqia, 2018-08-10:
@LarryYourWaiter hi，i am a beginner for this , i have a problem same like yours. what should i do if i want to output aircraft's GPS ? can u please give me some thoughts? thank u

LarryYourWaiter, 2018-08-10:
@souqia Before I can really help, I need some more information from you:
1. What do you mean by output? (sending to mobile, collecting in a log on the OSDK...) if you are sending to mobile, what OS is your mobile device running?
2. Do you have code already that attempts to "output" the data? If so, can you show me your code.
3. Do you have a specific kind of GPS data you want to receive? [Different types of GPS data scattered throughout this list](https://developer.dji.com/onboard-api-reference/namespaceDJI_1_1OSDK_1_1Telemetry.html)
4. What OS is your onboard computer using?

Some helpful tips (if you haven't already done them):
1. Read up on the [OSDK API reference](https://developer.dji.com/onboard-api-reference/index.html)
2. Trying to understand how telemetry and mobile samples in the OSDK work. These can be found in the [OSDK Github Repository](https://github.com/dji-sdk/Onboard-SDK). The only telemetry sample I found is for linux under Onboard-SDK/sample/linux/telemetry/telemetry_sample.cpp
3. For future questions post them on both [StackOverflow #dji-sdk](https://stackoverflow.com/questions/tagged/dji-sdk) and here since there is a greater chance you will solve your problem as DJI engineers help on both Github and StackOverflow. 

[NOTE] Im assuming you are able to run your program and only need help with your code. If not, say so.

LarryYourWaiter, 2018-08-11:
@souqia Is english your first language? If not, I do know a bit of mandarin.

souqia, 2018-08-13:
@LarryYourWaiter 老哥，你懂普通话真是6了。我就是想从M100无人机中输出飞机的经纬度信息（通过妙算manifold的USB口传输出来都可以）。现在我手上有大疆m100，和一个妙算manifold，因为刚开始弄这个，不知如何着手开始。我还没有代码。看了一下你说的文档，利用DJI::OSDK::Telemetry::GlobalPosition Struct和DJI::OSDK::DataBroadcast Class 看起来跟我要解决的问题相关性比较大，是不是这样？请给点思路，如果有示例代码更好，非常感谢。

imkhairulikhwan, 2018-08-13:
@jonathanmichel Sorry I am not good in C++ as well :/ I think I missed some class declarations or something when trying to setup the Linux example to send to mobile, that you provided:

Do I need to import other libraries in the file? 

Thanks! 

[image]

jonathanmichel, 2018-08-13:
@imkhairulikhwan Yes you have to include "dji_vehicle.cpp" and add `using namespace std`

imkhairulikhwan, 2018-08-13:
@jonathanmichel thanks for the prompt reply. 

I did include .hpp though because .cpp does not exist in the sample folder. Still receive the same error. I am guessing we need to create .cpp separately?

[image]

I hope this discussion does not spam the others, or perhaps we could take it in other channel if it's okay to you? :/

imkhairulikhwan, 2018-08-13:
@jonathanmichel on the sidenote, do you use different app Id when running Android Mobile SDK and for Linux OnBoard SDK? I'm guessing yes, but just want to confirm on that.

LarryYourWaiter, 2018-08-13:
@souqia 如果你不知道这个好，你因该看着所有DJI Documentation。那个比Github好。我没有manifold，所以我不会说话关于那。祝你好运。如果你有问题，我在这。对不起我的中文不好。

jonathanmichel, 2018-08-13:
@imkhairulikhwan Sorry I made a mistake in my last post. I thought "dji_vehicle.hpp" and not .cpp of course. 

It's hard to help you but I think you have a problem with your project structure. Take a look at this [gist](https://gist.github.com/jonathanmichel/3a1fffec07e71a02dcc11c70a02d9d58) and leave a comment if needed (to avoid spam here). I'm not totaly sure it compiles because I don't have a Linux on hand but it should. 

Concerning the app Key, with the Mobile SDK you don't need an app Id. You have to set a Bundle Identifier (com.mycompagny.myproject) and it gives you a app key. With Onboard SDK an Id and a key are generated and they have to be used in UserConfig.txt

imkhairulikhwan, 2018-08-14:
@jonathanmichel Thanks for the thorough details and samples! I have left some comments inside it.

Noted on the Mobile SDK as well.

imkhairulikhwan, 2018-08-14:
I tried to run both Linux and ROS samples (not at the same time), but still failed to get the command from the mobile sdk. I use Android Kotlin to send the command. 

[image]

imkhairulikhwan, 2018-08-14:
@LarryYourWaiter did you try the sample provided? I did try iOS sample from https://github.com/dji-sdk/Mobile-OSDK-iOS-App as well but it seems not to be working (no response whatsoever)

souqia, 2018-08-14:
@LarryYourWaiter 首先很感谢你能回复，而且你说的建议，我都看了，也阅读了onboardsdk文档。你中文很棒，可以加一下你的邮箱或者其他联系方式吗？请教一些问题

LarryYourWaiter, 2018-08-14:
@souqia email: jforteemail@gmail.com

imkhairulikhwan, 2018-08-21:
Hi @LarryYourWaiter , 

may I know if you have been able to run the two-way communication between Mobile SDK and Onboard SDK? If not mistaken, from your previous post, you are using iOS to initiate the communication is that right?

Previously I am able to do two-way communication for Android but for iOS, only from mobile to onboard, but we could not receive the message from onboard.

Thank you in advance!

LarryYourWaiter, 2018-08-21:
@imkhairulikhwan yes I have both down and upstream functional. I do not communicate commands but data and messages, what do you need help with?

imkhairulikhwan, 2018-08-21:
@LarryYourWaiter, just now we faced issue to receive the message from onboard to ios. But we have found the culprit (by assigning the variable to nullable seems causing the issue).

benUrsel, 2018-08-21:
@imkhairulikhwan can you explain what you did to solve the issue? I haven't been able to recieve a message in iOS

imkhairulikhwan, 2018-08-27:
Sorry for the late reply, @benUrsel there's another person working on iOS Mobile SDK (written in Swift), I attach here the snippet code for your reference. Hope it helps.

[dji onboard.docx](https://github.com/dji-sdk/Onboard-SDK/files/2322337/dji.onboard.docx)

closed 2021-06-24 by dji-kyle