dji-sdk/Onboard-SDK #360: Microsoft SDK info

jasperza, 2018-08-06:
Apologies if this is the wrong forum but does anyone know where we can get the Microsoft SDK from that has been promised for a while?

imkhairulikhwan, 2018-08-07:
@jasperza you meant the SDK for Windows 10 right?
I am awaiting the public release for that as well. I have emailed the support previously, but they did not have the exact date either. It says here it should be this fall. 

But, hopefully someone from DJI could provide the exact date for that. 

https://www.windowscentral.com/dji-and-microsoft-team-sdk-windows-10

jasperza, 2018-08-07:
Yes! Phew glad it's not just me missing out in the southern hemisphere. It would be good if us subscribers to this technical forum could get an early release for evaluation, I think anyone reading this is technically qualified to handle it!

imkhairulikhwan, 2018-08-07:
True! An early release would help both customer and DJI sides to get to know on the features available. 

I believe this forum is the place dedicated for such discussion as well. If not mistaken, DJI announced a while back on the direction to focus on the Github forum. :D

dji-lisafedane, 2018-08-30:
DJI is moving our SDK community to Stack Overflow with tag #dji-sdk.  If you would like a preview of the Window's SDK email dev@dji.com with your request.  And yes, the ETA for the official release is not yet determined but we are shooting for fall.  I'm closing this thread now but please feel free to email us or go to Stack Overflow

closed 2018-08-30 by dji-lisafedane