dji-sdk/Onboard-SDK #365: Guidance SDK deprecated, says to use Onboard SDK

adam3039, 2018-08-20:
I recently purchased Guidance for my M100. On the Guidance website below, it says that the Guidance SDK has been deprecated, and to use the Onboard SDK to achieve similar results.

https://developer.dji.com/guidance-sdk/

I'm using the Onboard SDK plenty right now for flight control stuff, but it doesn't seem like any of the Guidance functionality has been brought over to the Onboard SDK. I'm hesitant to develop anything using the Guidance SDK when it is deprecated. 

Is there any official word from DJI on this decision? I haven't been able to find anything.

Thanks.

dangviethieu-hntv, 2018-08-22:
@adam3039 I see in "Onboard-SDK/osdk-core/api/inc/dji_telemetry.hpp" have:
typedef struct RelativePosition
{
  float32_t down;            /*!< distance from obstacle (cm) */
  float32_t front;           /*!< distance from obstacle (cm) */
  float32_t right;           /*!< distance from obstacle (cm) */
  float32_t back;            /*!< distance from obstacle (cm) */
  float32_t left;            /*!< distance from obstacle (cm) */
  float32_t up;              /*!< distance from obstacle (cm) */
  uint8_t   downHealth : 1;  /*!< Down sensor flag: 0 - not working, 1 - working */
  uint8_t   frontHealth : 1; /*!< Front sensor flag: 0 - not working, 1 - working */
  uint8_t   rightHealth : 1; /*!< Right sensor flag: 0 - not working, 1 - working */
  uint8_t   backHealth : 1;  /*!< Back sensor flag: 0 - not working, 1 - working */
  uint8_t   leftHealth : 1;  /*!< Left sensor flag: 0 - not working, 1 - working */
  uint8_t   upHealth : 1;    /*!< Up sensor health flag: 0 - not working, 1 - working */
  uint8_t   reserved : 2;    /*!< Reserved sensor health flag*/
} RelativePosition;
I wonder, perhaps DJI will use it in order to replace Guidance in future ?

dji-lisafedane, 2018-08-23:
Please continue to use the Guidance SDK to develop for your Guidance hardware - we are leaving the GSDK open for this very purpose.  The deprecation of the GSDK only means that this particular SDK will no longer be receiving any updates.  Since guidance functionality is now integrated in the M200 drone series, the **_OSDK_** does not support support development on Guidance hardware.  

We apologize that the alert on the GSDK front page did not clarify this and thank you for bringing this to our attention as we're now in the process of updating the message.

jasperza, 2018-08-23:
Lisa thanks for your response , we have built a business on m100 with guidance, we don't use the sdk, but we need assurance that the guidance hardware will not be depracated. The M200 range does not have all the features and flexibility of guidance.

dji-lisafedane, 2018-08-23:
@jasperza I see the legitimacy of your concern and please reach out to DJI's main support line at support@dji.com to ask about this. Only SDK engineers monitor this GitHub forum and unfortunately can not answer hardware questions.

closed 2018-08-30 by dji-lisafedane