dji-sdk/Onboard-SDK #367: New Device Error

rakshith95, 2018-08-29:
I get this error log when I try to run the camera-stream-callback-sample.  The drone (210RTK) was initialized earlier. I generated an App ID and app key through this link : https://developer.dji.com/user/apps/
Screenshot: https://imgur.com/a/dkoUHjy
The error Log: 
Please enter the type of camera stream you want to view
m: Main Camera
f: FPV  Camera
f
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = 095XE9X00200VB

STATUS/1 @ parseDroneVersionInfo, L708: Hardware = PM410

STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.3.10.4

ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!

STATUS/1 @ init, L49: Looking for USB device...

STATUS/1 @ init, L65: Found 7 USB devices, identifying DJI device...

STATUS/1 @ init, L83: Found a DJI device...

STATUS/1 @ init, L96: Attempting to open DJI USB device...

STATUS/1 @ init, L134: ...DJI USB device started successfully.

STATUS/1 @ activate, L1277: version 0x3030A04

STATUS/1 @ getErrorCodeMessage, L656: activate
STATUS/1 @ getCMDSetActivationMSG, L705: NEW_DEVICE_ERROR

ERRORLOG/1 @ activate, L1302: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

ERRORLOG/1 @ activate, L1309: Solutions for NEW_DEVICE_ERROR:
* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
* If this is a new device, please make sure your DJI Go App is connected to internet to activate the new device for the first time.
* If this device is previously activated with another app_id and app_key, you will need to re-activate it again (with internet through DJI GO App).
* A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.

STATUS/1 @ getErrorCodeMessage, L656: initVehicle
STATUS/1 @ getCMDSetActivationMSG, L705: NEW_DEVICE_ERROR

STATUS/1 @ getErrorCodeMessage, L656: reset
STATUS/1 @ getCMDSetSubscribeMSG, L739: SUBSCRIBER_ILLEGAL_DATA_LENGTH
Vehicle not initialized, exiting.

imkhairulikhwan, 2018-08-29:
I think you should not reveal your app key and ID as it's supposed to be kept internally.

Anyhow, have you tried as per error message said? 

- Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
- If this is a new device, please make sure your DJI Go App is connected to internet to activate the new device for the first time.
- If this device is previously activated with another app_id and app_key, you will need to re-activate it again (with internet through DJI GO App).
- A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice

rakshith95, 2018-08-29:
I have used  flown the device with the controller and the DJI GO4 app before, so it is not a new device. However, I created a dev account and generated the ID, and key only a couple of days back. Is there anything else I'm missing?

imkhairulikhwan, 2018-08-29:
I see. If that's the case, you are using the new app ID and key at the moment, which as advised from the error message, you will have to try to activate it twice. If you already did that, I am not sure of what could cause the error then :/

rakshith95, 2018-08-29:
I found the error. The Go4 app wasn't connected to the drone controller.

closed 2018-08-29 by rakshith95

pointgeo, 2018-10-03:
We have a similar problem, but with M600 Pro and Raspberry Pi 3 w/Raspbian :

****************
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyS0 with baudrate 115200...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = xxxxxxxxx

STATUS/1 @ parseDroneVersionInfo, L708: Hardware = xxxxxxxxxx

STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.2.41.13

ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1277: version 0x302290D

STATUS/1 @ getErrorCodeMessage, L656: activate
STATUS/1 @ getCMDSetActivationMSG, L705: NEW_DEVICE_ERROR

ERRORLOG/1 @ activate, L1302: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

ERRORLOG/1 @ activate, L1309: Solutions for NEW_DEVICE_ERROR:
	* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
	* If this is a new device, please make sure your DJI Go App is connected to internet to activate the new device for the first time.
	* If this device is previously activated with another app_id and app_key, you will need to re-activate it again (with internet through DJI GO App).
	* A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.

STATUS/1 @ getErrorCodeMessage, L656: initVehicle
STATUS/1 @ getCMDSetActivationMSG, L705: NEW_DEVICE_ERROR

STATUS/1 @ getErrorCodeMessage, L656: reset
STATUS/1 @ getCMDSetSubscribeMSG, L739: SUBSCRIBER_ILLEGAL_DATA_LENGTH
Vehicle not initialized, exiting.
*****************
Using GO app on iPad, and the "authorized question" pops up. No difference when authorizing.

Please help.

man2607, 2019-10-26:
Hey, I'm having the same issue. Could you please elaborate on how you fixed it? Thanks.