dji-sdk/Onboard-SDK #368: camera-stream-callback-sample

rakshith95, 2018-08-29:
I'm trying to run the camera-stream-callback-sample, and when I choose the option for main camera, this is the output I am getting: 
lease enter the type of camera stream you want to view
m: Main Camera
f: FPV  Camera
m 
Read App ID
User Configuration read successfully. 
STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = 095XE9X00200VB
STATUS/1 @ parseDroneVersionInfo, L708: Hardware = PM410
STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.3.10.4

ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!

STATUS/1 @ init, L49: Looking for USB device...
STATUS/1 @ init, L65: Found 8 USB devices, identifying DJI device...
STATUS/1 @ init, L83: Found a DJI device...
STATUS/1 @ init, L96: Attempting to open DJI USB device...
STATUS/1 @ init, L134: ...DJI USB device started successfully.
STATUS/1 @ activate, L1277: version 0x3030A04
STATUS/1 @ activate, L1290: Activation successful
STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L869: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.
STATUS/1 @ removePackage, L468: Remove package 0 successful.

STATUS/1 Connect to MAIN_CAMERA successful
STATUS/1 All components for decoding initialized ...
STATUS/1 
STATUS/1 ****** Decoder Callback Thread Start ******
STATUS/1 Decoder Callback Thread Stopped...
STATUS/1 User callback thread created successfully!
**** MAIN_CAMERA data reading thread start! ****
STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

STATUS/1 Connect to MAIN_CAMERA successful
STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

STATUS/1 Connect to MAIN_CAMERA successful
STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

STATUS/1 Connect to MAIN_CAMERA successful
STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

STATUS/1 Connect to MAIN_CAMERA successful
STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

STATUS/1 Connect to MAIN_CAMERA successful
STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

rakshith95, 2018-08-29:
When I choose the fpv camera, I get a different issue: 

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 Connect to FPV_CAMERA successful
STATUS/1 All components for decoding initialized ...
STATUS/1 **** FPV_CAMERA data reading thread start! ****
STATUS/1 ****** Decoder Callback Thread Start ******
STATUS/1 Decoder Callback Thread Stopped...
STATUS/1 User callback thread created successfully!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[... 126 lines cut ...]
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] non-existing PPS 63 referenced
[h264 @ 0x55acfa5c8ec0] decode_slice_header error
[h264 @ 0x55acfa5c8ec0] no frame!
Assertion desc failed at src/libswscale/swscale_internal.h:668
Aborted (core dumped)

ekieffer, 2018-08-29:
@rakshith95 

We get exactly the same issue for the FPV camera.  

Looking forward to get a fix.

rakshith95, 2018-08-29:
I didn't read the part about having the RC on before I ran it for the main camera. But yeah, I'm getting the same output for that as well.

zliDJI, 2018-08-29:
Hi @rakshith95, about your issue/screen shot with FPV camera, it looks like something related to the ffmpeg decoder package. Could you try to run the sample on a Ubuntu 16.04 machine? Our code was not tested on Ubuntu 18.

about the main camera issue in your post (lost connection and reconnect), I guess you ran it without RC on, right?  If you have RC on and run sample on main camera, you should get the same ffmpeg decoding error as the FPV camera case.

rakshith95, 2018-08-30:
@zliDJI I am able to get the stream from the stereo-vision-single-thread-sample though. If it was a problem native to 18.04, wouldn't that cause a problem as well?

rakshith95, 2018-08-30:
Running it on an Ubuntu 16.04 system, I am able to get the feed from the main camera. Hope this is sorted out for 18,04 soon.

closed 2018-08-30 by rakshith95

alejodosr, 2019-05-23:
Any news related to Ubuntu 18.04 and ffmpeg? Same issue here.

alejodosr, 2019-05-30:
I have managed to solve it. The problem was related to FFmpeg library. I have downloaded the source code, changed it to `release/2.8` and linked osdk to this version. If interested, here you have an adapted version of both [Onboard-SDK ](https://github.com/alejodosr/Onboard-SDK) and [Onboard-SDK-ROS](https://github.com/alejodosr/Onboard-SDK-ROS). For building Onboard-SDK, please:

```
cd Onboard-SDK/FFmpeg
./configure
make
cd Onboard-SDK
mkdir build
cd build
cmake .. -DADVANCED_SENSING=ON
make
```

Now, the `camera-stream-callback-sample` should be working properly. Hope it helps.

prit-ubuntu, 2020-06-24:
@alejodosr this version works perfectly! thanks for helping!