dji-sdk/Onboard-SDK #369: Ubuntu - " non-existing PPS 0 referenced" "decode_slice_header error" on M210 using OSDK

ekieffer, 2018-08-29:
Hi,

We are trying to execute the "advanced sensing" examples on a matrice 210 according to the DJI Onboard SDK guidelines.

We installed all the required libraries on a Ubuntu 18.04 LTS.  When starting the  "camera-stream-poll-sample", we get the following errors:

> h264 @ 0x55b57408a660] non-existing PPS 0 referenced
> [h264 @ 0x55b57408a660] decode_slice_header error
> [h264 @ 0x55b57408a660] no frame!

Could you please tell us how to fix this issue ?

Thank you

Current configuration:
onboard os: ubuntu 18.04
matrice 210 firmware: latest (3.3.10.4)

zliDJI, 2018-08-30:
See #368 and try to run the code on an Ubuntu 16.04 machine. Please kindly report your results, thanks!

ekieffer, 2018-08-31:
Hi @zliDJI,

So on Ubuntu 16.04, it works. In fact, "libswresample2" is the problem on Ubuntu 18.04.  Hope this info will help you fixing it for the future Ubuntu releases.

We have an additional question regarding the advanced-sensing library.  We noticed that the branches "advanced-sensing-2.0.1-armv7" does not support the camera frame API for advanced sensing.  Are you planning to work on it ? 

Thank you

alejodosr, 2019-05-28:
Hi @zliDJI , we are having the same issue with Ubuntu 18, and since we are required to use Nvidia Jetson AGX Xavier, we are forced to solved this issue. 

Did you manage to figure out anything else? How did you come to the conclusion that the problem was *libswresample2*? Thank you in advance.

alejodosr, 2019-05-30:
I have managed to solve it. The problem was related to FFmpeg library. I have downloaded the source code, changed it to `release/2.8` and linked osdk to this version. If interested, here you have an adapted version of both [Onboard-SDK ](https://github.com/alejodosr/Onboard-SDK) and [Onboard-SDK-ROS](https://github.com/alejodosr/Onboard-SDK-ROS). For building Onboard-SDK, please:

```
cd Onboard-SDK/FFmpeg
./configure
make
cd Onboard-SDK
mkdir build
cd build
cmake .. -DADVANCED_SENSING=ON
make
```

Now, the `camera-stream-callback-sample` should be working properly. Hope it helps.

guischmitd, 2020-05-28:
@alejodosr did you have any success getting the main camera stream on ubuntu 18.04 since this post? Were you able to trace any of the problems that prevent getting the main camera stream?

@dji-dev any news on ubuntu 18.04 support for advanced sensing? Recent nvidia SOCs (Xavier, Nano) don't even have official support for 16.04. It's an old OS, and it honestly makes me uneasy that there is no support whatsoever for 18.04 advanced sensing at this time.

alejodosr, 2020-06-08:
@guischmitd the key was ffmpeg. What I did was to build ffmpeg from source and linked locally to the osdk. With that, I was able to retrieve images from the osdk and osdk-ros

guischmitd, 2020-06-08:
@alejodosr I see, so, just to confirm: you could get frames from the X4S as well by compiling ffmpeg from source? thanks for the replies so far, your forked repo has been a life saver!

alejodosr, 2020-06-08:
@guischmitd yes, I confirm, I could receive images from X4S both with OSDK and OSDK-ROS

prit-ubuntu, 2020-06-13:
@alejodosr I'm planning to use the relatively new NVIDIA Xavier for advanced sensing with OSDK. Do you have a similar system that works for this? Because I see that Xavier doesn't allow you to use Ubuntu 16.04...

alejodosr, 2020-06-15:
@prit-ubuntu The instructions from https://github.com/dji-sdk/Onboard-SDK/issues/369#issuecomment-497295663 are compatible to Xavier, since I was exactly using that system too.

prit-ubuntu, 2020-06-23:
@alejodosr thank you. I have 2 follow up questions: 1. Were you able to use DJI-SDK-ROS, since on the developer's page, it says that it is only compatible with ROS Kinetic, whereas XAVIER does not have Kinetic support?
2. Did you install Jetpack before starting to run DJI-SDK on the Xavier module? I'm not sure if the Linux scripts have support for L4T which is a part of jetpack..

alejodosr, 2020-06-26:
@prit-ubuntu  I answer you below

1. Yes, I [forked it](https://github.com/alejodosr/Onboard-SDK-ROS) and it is fully compatible with ROS Melodic.
2. I was using Xavier and normal jetpack installation. The level of abstraction of Ubuntu 18 was enough. In the end, installing DJI-SDK and DJI-SDK-ROS involves building c++ code and running it, and that is fully compatible with the system.

beyond21299, 2020-07-12:
I tried to use the advanced vision module to collect the video stream of Z30 camera, but the image was a flowered screen. The error code is as follows:

dji-kyle, 2021-06-24:
The video format of Z30 is GDR, it can be decoded via param settings: -flags2 showall, and please ensure the firmware version has been updated to the latest.
eg:
./ffplay -flags2 showall -f h264 XXXXXXX.h264

closed 2021-06-24 by dji-kyle