dji-sdk/Onboard-SDK #374: auto start dji program when boot

Jai-GAY, 2018-09-13:
I am trying to boot my pi and run the [djiosdk-flightcontrol-sample](https://developer.dji.com/onboard-sdk/documentation/quick-start/quick-start.html#run-the-sample-app) at boot time using the following [Method 1: rc.local](https://www.dexterindustries.com/howto/run-a-program-on-your-raspberry-pi-at-startup/). My rc.local content:

sudo usermod -a -G dialout $USER &
sudo /home/pi/dji-sdk/Onboard-SDK-3.6/build/bin/djiosdk-flightcontrol-sample_boot /home/pi/dji-sdk/Onboard-SDK-3.6/build/bin/UserConfig2.txt 2>&1 |tee /home/pi/dji-sdk/Onboard-SDK-3.6/build/bin/startup_log.txt &

However, i get the following error which i don't get it when i use a terminal and enter the command lines.

Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/serial0 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
ERRORLOG/1 @ getDroneVersion, L1339: **Drone version not obtained**! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are inverted.
		* Serial port is occupied by another program.
		* Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

ERRORLOG/1 @ activate, L1228: Unable to initialize some vehicle components!
STATUS/1 @ getErrorCodeMessage, L644: initVehicle
STATUS/1 @ getCommonErrorCodeMessage, L710: ACK_NO_RESPONSE_ERROR
Vehicle not initialized, exiting.

Jai-GAY, 2018-09-14:
after further debug, it seems fail at LinuxSetup linuxEnvironment(argc, argv); if startup at boot time.
Same code does not happen using command line.

Jai-GAY, 2018-09-14:
Is it normal that i cannot see implementation for LinuxSetup linuxEnvironment(argc, argv); in DJI_linux_helpers.hpp and DJI_linux_helpers.cpp under \Onboard-SDK-3.6\sample\linux\common ?

class LinuxSetup

Jai-GAY, 2018-09-20:
[Method 5: crontab](https://www.dexterindustries.com/howto/run-a-program-on-your-raspberry-pi-at-startup/#crontab) also not working.

DarkTrick, 2019-04-19:
They change, because rawACK is a pointer in memory which changes at every rerun. 
You should use
DERROR ("LinuxSetup::getDroneVersion A = 0x%X\n", *rawACK);
instead (note the asterisk before rawACK).

Jai-GAY, 2019-06-07:
this finally works on my Pi3. missing **&** all along.

sudo EDITOR=gedit crontab -e

@reboot sleep 10 && /absolute-path-to-djiosdk-mobile-sample /absolute-path-to-UserConfig.txt >> /absolute-path-to-logfile 2>&1 **&**

closed 2019-06-07 by Jai-GAY