dji-sdk/Onboard-SDK #378: Lat, Lon are 0.0 in getGlobalPosition()

jarednm, 2018-09-25:
I am using DJI OSDK 3.7 on an M210 running latest firmware at time of writing (3.3.10.4) and have encountered an issue with the getGlobalPosition() function in broadcast.  

Though I am able to read a value for altitude, both the latitude and longitude are always reported as 0.0, even when latitude and longitude are nonzero in the getGPSInfo() message.  The reason I cannot simply use getGPSInfo() is because my application may be run on an M100, which does not support that getter.

I have verified that I have at least 4 satellites, and I gave the M210 up to 30 minutes to report latitude, longitude.

Attached are screenshots from my application, printing latitude and longitude from getGlobalPosition(), and getGPSInfo().  

[image]
[image]

huangyinpei, 2019-03-27:
hi. I want to know how does DJI Assistant2(simulation) CALCULATE worldlatitude, worldlongitude values? How did you calculate lata nd  lon .Please help me.thanks.

SChalice, 2019-06-25:
I am seeing the same thing on my M600. This is definitely a bug because the unpack routine has the data there, in the gps area, it was just not put into the gp area.

```
  unpackOne(FLAG_POSITION    ,&gp        ,pdata,sizeof(gp        ));
  unpackOne(FLAG_POSITION    ,&rp        ,pdata,sizeof(rp        ));
  unpackOne(FLAG_GPSINFO     ,&gps       ,pdata,sizeof(gps       ));
```

closed 2021-06-24 by dji-kyle