dji-sdk/Onboard-SDK #379: NEW_DEVICE_ERROR

pointgeo, 2018-10-03:
We have a Activationr problem with our M600 Pro and Raspberry Pi 3 w/Raspbian :

******************
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyS0 with baudrate 115200...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = xxxxxxxxx

STATUS/1 @ parseDroneVersionInfo, L708: Hardware = xxxxxxxxxx

STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.2.41.13

ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1277: version 0x302290D

STATUS/1 @ getErrorCodeMessage, L656: activate
STATUS/1 @ getCMDSetActivationMSG, L705: NEW_DEVICE_ERROR

ERRORLOG/1 @ activate, L1302: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

ERRORLOG/1 @ activate, L1309: Solutions for NEW_DEVICE_ERROR:
* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
* If this is a new device, please make sure your DJI Go App is connected to internet to activate the new device for the first time.
* If this device is previously activated with another app_id and app_key, you will need to re-activate it again (with internet through DJI GO App).
* A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.

STATUS/1 @ getErrorCodeMessage, L656: initVehicle
STATUS/1 @ getCMDSetActivationMSG, L705: NEW_DEVICE_ERROR

STATUS/1 @ getErrorCodeMessage, L656: reset
STATUS/1 @ getCMDSetSubscribeMSG, L739: SUBSCRIBER_ILLEGAL_DATA_LENGTH
Vehicle not initialized, exiting.
*************************************
Using GO app on iPad, and the "authorized question" pops up. No difference when authorizing.

Please help.

dwang0927, 2018-10-03:
Encountered the same problem, Using M100 with Raspberry Pi 3B w/Raspbian. 

Read App ID
User Configuration read successfully. 
STATUS/1 @ init, L55: Attempting to open device /dev/ttyS0 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = 0410200473
STATUS/1 @ parseDroneVersionInfo, L708: Hardware = M100
STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.1.10.0
STATUS/1 @ parseDroneVersionInfo, L712: Version CRC = 0xA6453AAC
STATUS/1 @ initSubscriber, L757: Telemetry subscription mechanism is not supported on this platform!
STATUS/1 @ initMFIO, L960: MFIO is not supported on this platform!
STATUS/1 @ initHardSync, L1041: Hardware Sync is not supported on this platform!
STATUS/1 @ activate, L1277: version 0x3010A00
STATUS/1 @ getErrorCodeMessage, L656: activate
STATUS/1 @ getCMDSetActivationMSG, L705: NEW_DEVICE_ERROR
ERRORLOG/1 @ activate, L1302: Failed to activate please retry SET 0x0 ID 0x1 code 0x3
ERRORLOG/1 @ activate, L1309: Solutions for NEW_DEVICE_ERROR:
	* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
	* If this is a new device, please make sure your DJI Go App is connected to internet to activate the new device for the first time.
	* If this device is previously activated with another app_id and app_key, you will need to re-activate it again (with internet through DJI GO App).
	* A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.
STATUS/1 @ getErrorCodeMessage, L656: initVehicle
STATUS/1 @ getCMDSetActivationMSG, L705: NEW_DEVICE_ERROR
Vehicle not initialized, exiting.

sdo91, 2018-10-03:
I am seeing this same issue of NEW_DEVICE_ERROR.
We are using a new M600 pro with sdk v3.6. Our older M600s still connect just fine.
We have connected to the DJI GO app (with internet) and approved the new device, but we still get the same error after multiple reboots.
I'm not sure what is meant by this phrase: "A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice." The DJI GO app only gives the option to activate once. Is there a different way to activate the new M600?

dwang0927, 2018-10-10:
Just got this problem solved by changing some configurations in DJI Assistant 2. In the settings, you need to enable User account information, DJI device serial number, and Onboard SDK APP ID.

nZac, 2020-02-10:
We fixed this with @dwang0927 suggestion. If you registered your app after you logged in and activated your drone the first time, log out and log back in again.

dji-kyle, 2021-06-24:
The following forum community is recommanded for you, if the problem persists, or new questions about the DJI SDK,
https://djisdksupport.zendesk.com/hc/en-us/community/topics

closed 2021-06-24 by dji-kyle

ehsan-kouchaki, 2025-03-27:
I solved this problem by simultaneously connecting the OSDK device and DJI Assistant 2 to the drone. You need to do this only once.