dji-sdk/Onboard-SDK #382: Virtual RC is not supported on this platform!

Flipped03, 2018-10-18:
when I use N3 to run the  flightcontrol sample，there is  no response on the simulator.  Fllowing are the log:

Read App ID
User Configuration read successfully. 
STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = 0DBDF5E00201GP
STATUS/1 @ parseDroneVersionInfo, L708: Hardware = N3
STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.3.8.39
ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1277: version 0x3030827

STATUS/1 @ activate, L1290: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L869: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L913: Gimbal not mounted!
| Available commands:                                            |
| [a] Monitored Takeoff + Landing                                |
| [b] Monitored Takeoff + Position Control + Landing             |

when I enter a ,then it becomes:
STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=10, nTopics=2.

STATUS/1 @ getErrorCodeMessage, L656: ����
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

STATUS/1 @ verify, L244: Verify subscription successful.
ERRORLOG/1 @ initPackageFromTopicList, L167: package [0] is being occupied.

STATUS/1 @ reset, L567: Reset Subscription Successful.

How can I to run the sample with N3 in simulator.

michaelumo, 2018-12-05:
Hi, I'm having same kind of problem.
I tried two different ways to run flightcontrol-sample.

-WAY1:
When I turn "on" Futaba RC, turn on the N3, run ./djiosdk-flightcontrol-sample, the following error occurs.

-error1-----------------------------------------------------------------------------------------
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 115200...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = 0DBDF3800200QP

STATUS/1 @ parseDroneVersionInfo, L708: Hardware = N3

STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.3.8.39

STATUS/1 @ removePackage, L468: Remove package 0 successful.
ERRORLOG/1 @ removeLeftOverPackages, L499: failed to remove package 0
ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1277: version 0x3030827

STATUS/1 @ activate, L1290: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L869: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L913: Gimbal not mounted!
| Available commands: |
| [a] Monitored Takeoff + Landing |
| [b] Monitored Takeoff + Position Control + Landing |
------------------------------------------------------------------------------------------------

In this case when I press 'a', it displays the next error.

-error2-----------------------------------------------------------------------------------------
STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=10, nTopics=2.

STATUS/1 @ getErrorCodeMessage, L656: 
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE
------------------------------------------------------------------------------------------------

Pressing 'a' one more time, and it displays this error.

-error3-----------------------------------------------------------------------------------------
STATUS/1 @ verify, L244: Verify subscription successful.
ERRORLOG/1 @ initPackageFromTopicList, L167: package [0] is being occupied.
------------------------------------------------------------------------------------------------

The drone does nothing in the simulator.

-WAY2:
When I turn "off" Futaba RC, turn on the N3, run ./djiosdk-flightcontrol-sample, error1 occurs.
Pressing 'a' displays this.

------------------------------------------------------------------------------------------------
STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=10, nTopics=2.
Motors spinning...
Ascending...
Successful takeoff!

STATUS/1 @ removePackage, L468: Remove package 0 successful.

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=10, nTopics=2.
Landing...
Successful landing!
------------------------------------------------------------------------------------------------

The drone takes off and lands in the simulator.

-System Information
Hardware:
	raspberry pi 3 model B
	dji N3 Flight Controller
	Futaba 2.4GHz FASSTest T14SG (using SBUS)

OS:
	NAME="Ubuntu"
	VERSION="16.04.2 LTS (Xenial Xerus)"
	ID=ubuntu
	ID_LIKE=debian
	PRETTY_NAME="Ubuntu 16.04.2 LTS"
	VERSION_ID="16.04"

N3:
	Firmware version: V1.7.6.0
	Failsafe Action: Hover
	Module Status: all checked

DJI Assistant:
	Version: V1.2.4

Thanks.

michaelumo, 2019-01-11:
Solved it by shifting the throttle stick on the remote controller to zero.

AbhishekGS, 2019-06-14:
@Flipped03, I am facing the same error. Were you able to solve this? 
I am trying DJI N3 with Onboard SDK for the first time. I do not have the rc connected to the SBUS port of the DJI N3. 

Is RC necessary for using the Simulator from DJI assistant 2, for running flightcontrol sample?

closed 2021-06-24 by dji-kyle