dji-sdk/Onboard-SDK #384: How to modify flightcontrolsample to fly to different points

coldshock1123, 2018-10-19:
Hi,

The stock DJI OSDK 3.7 came with several flight_control_samples.cpp and main.cpp locations such as one in sample>linux>flight-control and another in build>bin>flight-control-sample. I tried modifying the one in sample>linux>flight-control however the drone did not react to any changes. May i know which is the right file to work on ? The OSDK on DJI documentation does not mention which file thus i'm really lost as i'm new to SDKs too. Please help me! 

I'm running ubuntu and M100 with Nvidia TX2, samples can run however i need to modify them.

Thank you in advance!

Jai-GAY, 2018-10-19:
it could be waiting for your user keyboard input.
std::cin >> inputChar;

try change to 
#if 1
  inputChar = 'b';
  usleep (1000000);
  std::cout << "inputchar = " << inputChar << "\n\n";
#else
  std::cin >> inputChar;
#endif

coldshock1123, 2018-10-19:
Hi thanks for responding :). Is this what you mean? Replacing "std::cin >> inputChar;" with 
"#if 1
inputChar = 'b';
usleep (1000000);
std::cout << "inputchar = " << inputChar << "\n\n";
#else
std::cin >> inputChar;
#endif" 

Heres the sample i ran however the drone still flew the exact default sample flightpath despite my changes to the Offset. 

#include "flight_control_sample.hpp"

using namespace DJI::OSDK;
using namespace DJI::OSDK::Telemetry;

/*! main
 *
 */
int
main(int argc, char** argv)
{
  // Initialize variables
  int functionTimeout = 1;

  // Setup OSDK.
  LinuxSetup linuxEnvironment(argc, argv);
  Vehicle*   vehicle = linuxEnvironment.getVehicle();
  if (vehicle == NULL)
  {
    std::cout << "Vehicle not initialized, exiting.\n";
    return -1;
  }

  // Obtain Control Authority
  vehicle->obtainCtrlAuthority(functionTimeout);

  // Display interactive prompt
  std::cout
    << "| Available commands:                                            |"
    << std::endl;
  std::cout
    << "| [a] Monitored Takeoff + Landing                                |"
    << std::endl;
  std::cout
    << "| [b] Monitored Takeoff + Position Control + Landing             |"
    << std::endl;
  char inputChar;
 #if 1
inputChar = 'b';
usleep (1000000);
std::cout << "inputchar = " << inputChar << "\n\n";
#else
std::cin >> inputChar;
#endif

  switch (inputChar)
  {
    case 'a':
      monitoredTakeoff(vehicle);
      monitoredLanding(vehicle);
      break;
    case 'b':
      monitoredTakeoff(vehicle);
      moveByPositionOffset(vehicle, 60, 60,100, 30);
      moveByPositionOffset(vehicle, 0, 0, 100, -30);
      moveByPositionOffset(vehicle, 3, 3, 0, 0);
      monitoredLanding(vehicle);
      break;
    default:
      break;
  }

  return 0;
}

Jai-GAY, 2018-10-19:
the parameters are in meter, try smaller value. where you run your OSDK application, raspberry Pi ?

are you using simulator ? also your [Remote controller are sensitive](https://github.com/dji-sdk/Onboard-SDK/issues/344) and make sure UA is calibrated.

double check date time of your ..\build\bin files are correct and updated.

i had try adding my function into it and works. but i have not try changing the moveByPositionOffset values yet.

coldshock1123, 2018-10-19:
Have changed the values to a reasonable ones :

      moveByPositionOffset(vehicle, 10, 10,10, 30);
      moveByPositionOffset(vehicle, 10, 30, 10, -30);
      moveByPositionOffset(vehicle, 3, 3, 0, 0);
However, the simulator is still flying the default flight path of : 
      moveByPositionOffset(vehicle, 0, 6, 6, 30);
      moveByPositionOffset(vehicle, 6, 0, -3, -30);
      moveByPositionOffset(vehicle, -6, -6, 0, 0);
. Yes I'm using the simulator on Assistant 2 and i've checked that it have been calibrated with updated build bin files. I'm facing similar problems as you where it does not react to the modified values of the the moveByPositionOffset values

Jai-GAY, 2018-10-19:
So now i am not sure is Assistance 2 Simulator contains bug or the example is not working.

i am using [DJI Assistant 2 For Autopilot v2.0.0](https://www.dji.com/a3/info#downloads)

coldshock1123, 2018-10-20:
Same here, have reported to them but yet to receive a response :(. Thanks for your assistance tho!

Jai-GAY, 2018-10-25:
DJI support has [reproduced my issue](https://github.com/dji-sdk/Onboard-SDK/issues/384#issuecomment-431261199) and the fix works for my case by adding [vehicle->obtainCtrlAuthority(20); before line 548](https://github.com/dji-sdk/Onboard-SDK/blob/d43cf0ab135b015145ab7a2e41b85cbee6522f83/sample/linux/mobile/mobile_sample.cpp#L548)

Jai-GAY, 2018-11-01:
i am using Pi 3 to learn the SDK. i followed [this tutorial](https://developer.dji.com/onboard-sdk/documentation/development-workflow/environment-setup.html#ubuntu-linux). Not sure is it going to be helpful to you.

closed 2021-06-24 by dji-kyle