dji-sdk/Onboard-SDK #386: Drone version cannot be obtained

shehancaldera, 2018-10-23:
I'm trying to run the flight control sample aplication using OSDK 3.7 on a Ubuntu 16.04 Onboard PC. My drone is the Matrice 210 I have followed all the steps according to the documenttion but received the following error output:

```
scalder@ubuntu1:~/dji_SDKs/OSDK_37/Onboard-SDK/build/bin$ ./djiosdk-flightcontrol-sample UserConfig.txt 
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 115200...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1377: Drone version not obtained! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are inverted.
		* Serial port is occupied by another program.
		* Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login againscalder@ubuntu1:~/dji_SDKs/OSDK_37/Onboard-SDK/build/bin$ ./djiosdk-flightcontrol-sample UserConfig.txt 
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 115200...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1377: Drone version not obtained! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are inverted.
		* Serial port is occupied by another program.
		* Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

ERRORLOG/1 @ activate, L1266: Unable to initialize some vehicle components!
STATUS/1 @ getErrorCodeMessage, L656: initVehicle
STATUS/1 @ getCommonErrorCodeMessage, L722: ACK_NO_RESPONSE_ERROR
Vehicle not initialized, exiting.

ERRORLOG/1 @ activate, L1266: Unable to initialize some vehicle components!
STATUS/1 @ getErrorCodeMessage, L656: initVehicle
STATUS/1 @ getCommonErrorCodeMessage, L722: ACK_NO_RESPONSE_ERROR
Vehicle not initialized, exiting.
```
Would appreciate your support in resolving this issue. Thanks a lot!

Jai-GAY, 2018-10-23:
you could be [facing same error](https://github.com/dji-sdk/Onboard-SDK/issues/374#issuecomment-421901219) as me.

the Drone version acquired keep changing even on identical hardware, software and firmware.

shehancaldera, 2018-10-24:
Glad that I'm not the only one. Has there any fix being made?

Jai-GAY, 2018-10-25:
No.
did you try modify code to add debug print is it the same issue as mine ?

mine a bit different to yours as mine is working ok under command line execution, face problem when i wanted the board to auto start up after power on without login

shehancaldera, 2018-10-25:
Could you elaborate on debug print part? 

I'm trying to run it via command line execution. Can you explain how did you work around this issue when executing via command line? And also, is your osdk version is 3.7 or below? Thanks.

Jai-GAY, 2018-10-25:
i add [this line of code into the DJI example](https://github.com/dji-sdk/Onboard-SDK/issues/374#issuecomment-421901219).
DERROR ("LinuxSetup::getDroneVersion A = 0x%X\n", rawACK);

i didn't face this issue initially using command line. have [you calibrated both the FC and RC] ?(https://github.com/dji-sdk/Onboard-SDK/issues/344#issuecomment-412708695) ?

shehancaldera, 2018-10-25:
Thanks. I will check this. Tx and Rx pins are correctly connected. I didn't do any RC and FC calibration. It wasn't documented in OSDK documentation. And my hardware setup is the Dji M210 and a Ubuntu 16.04 pc. Remote controller is connected to the drone and Dji go is running. What followed was the hardware setup guide for M210 in OSDK documentation.

CptDD, 2019-02-05:
Hello,
Did you manage to solve this problem,because I am having the same issues and would be interested in a solution?

xelmirage, 2019-03-08:
guys, I just encountered the same situation and solved the problem in my case. I checked the com port of my manifold and my M600 and they are all working well. Finally I found out that since it says 

> L1266: Unable to initialize some vehicle components!

why not try mounting a gimbal on the drone? And after mounting the drone the error disappeared and the sample worked well. So the SDK is dumb and can't work without a gimal.

Hope this help:)

changmsdn, 2019-03-22:
Hi guys, this may be the cause: 

you have run the sample in a Linux way when there is a ROS master node running, and they conflict. You can't run `./djiosdk-flightcontrol-sample UserConfig.txt` and run `roslaunch dji_sdk sdk.launch` at the same time.

Just close the master ROS node and simply try the Linux way.

potBagMeat, 2019-08-04:
Did anybody solved this problem? I used Matrice 210 and Manifold 2 and encountered the same problem...
roslaunch dji_sdk sdk.launch
... logging to /home/dji/.ros/log/ea412c86-b684-11e9-9aa5-60601fce9b6d/roslaunch-manifold2-14664.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://manifold2:43959/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1076728
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: 5ee1c5614a147f80c...
 * /dji_sdk/serial_name: /dev/ttyACM0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

ROS_MASTER_URI=http://localhost:11311

process[dji_sdk-1]: started with pid [14681]
[ INFO] [1564904028.882658364]: Advanced Sensing is Enabled on M210.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyACM0 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1418: Drone version not obtained! Please do not proceed.
Possible reasons:
	Serial port connection:
		* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
		* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
		* TX and RX pins are inverted.
		* Serial port is occupied by another program.
		* Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

[ERROR] [1564904034.886974641]: drone activation error
[ERROR] [1564904034.887259288]: Vehicle initialization failed
^C[dji_sdk-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Jai-GAY, 2019-08-20:
DJI [has been denying it](https://github.com/dji-sdk/Onboard-SDK/issues/447) for quite some time.

dji-kyle, 2021-06-24:
The following forum community is recommanded for you, if the problem persists or new questions about the DJI SDK,
https://djisdksupport.zendesk.com/hc/en-us/community/topics

closed 2021-06-24 by dji-kyle

abisaz, 2022-10-07:
Yes, same her with the following error:

Read App ID
User Configuration read successfully.

[2989171.465]STATUS/1 @ getDroneVersion, L1702: ret = 255
[2989171.465]ERRORLOG/1 @ getDroneVersion, L1707: Drone version not obtained! Please do not proceed.
Possible reasons:
  Serial port connection:
        * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
        * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
        * TX and RX pins are invert
[2989171.466]STATUS/1 @ functionalSetUp, L282: Shake hand with drone Fail ! Cannot get drone version. (1/20)
[2989171.466]STATUS/1 @ functionalSetUp, L284: Try again after 1 second ......

My setup is the DJI Matrice 300, with a OSDK adapter board, a USB connection from it to an RPI3, as well as GND, TX, RX connected to the GPIO pins