dji-sdk/Onboard-SDK #387: Hotpoint mission - Change radius

jonathanmichel, 2018-11-01:
Hello,

I'm currently trying to change hotpoint mission radius during process. It works but I have a strange behavior.

When I use the [blocking call of `updateRadius`](https://github.com/dji-sdk/Onboard-SDK/blob/d43cf0ab135b015145ab7a2e41b85cbee6522f83/osdk-core/api/inc/dji_hotpoint.hpp#L203) it returns an error. If I get the error code it seems to be my last ack (in my case 'OBTAIN_CONTROL_SUCCESS`) and not the currently tested. 

With the [non-blocking call version of `updateRadius`](https://github.com/dji-sdk/Onboard-SDK/blob/d43cf0ab135b015145ab7a2e41b85cbee6522f83/osdk-core/api/inc/dji_hotpoint.hpp#L194), the callback is never called.

Resulting of this behavior, I can only update 30 times the radius of the mission before the dji_open_procol [indicates an error](https://github.com/dji-sdk/Onboard-SDK/blob/d43cf0ab135b015145ab7a2e41b85cbee6522f83/osdk-core/protocol/src/dji_open_protocol.cpp#L347) given that it allocates to many session waiting for ack. The communication with ack request is then completely locked.

Is that part (ack on hotpoint mission radius change) not yet implemented on the aircraft side ?

**Code**
```cpp
DJI::OSDK::HotPointSettings hpS = vehicle->missionManager->hpMission->getData();
auto radius = (float)hpS.radius;

int cnt = 1;
while (true) {
    if (radius <= 15) {
        radius += delta;

        DSTATUS("cnt=%d", cnt++);
        ACK::ErrorCode ack = vehicle->missionManager->hpMission->updateRadius(radius, 1);
        if (ACK::getError(ack) != ACK::SUCCESS) {
            DERROR("Update radius failed");
            ACK::getErrorCodeMessage(ack, __func__);
        }
        delay_ms(DELTA_UPDATE_MS);
    } else {
        LSTATUS("Radius limit of spiral mission reached")
        h->stop();
        break;
    }
}
```

**Log**
```
STATUS/1 @ addHotpoint, L199: New hotpoint set (Lon Lat Hei): 0.128165 	0.806633 	13.419594
STATUS/1 @ start, L106: Start hotpoint mission
STATUS/1 @ printInfo, L204: Mission Manager status: 

STATUS/1 @ printInfo, L205: There are 0 waypt missions and 1 hotpoint missions

STATUS/1 @ hpThread, L265: cnt=1
STATUS/1 @ start, L124: Start hotpoint mission successfully
ERRORLOG/1 @ hpThread, L262: Update radius failed
STATUS/1 @ getErrorCodeMessage, L656: hpThread
STATUS/1 @ getCMDIDSetControlMSG, L801: OBTAIN_CONTROL_SUCCESS

STATUS/1 @ hpThread, L265: cnt=2
ERRORLOG/1 @ hpThread, L262: Update radius failed
STATUS/1 @ getErrorCodeMessage, L656: hpThread
STATUS/1 @ getCMDIDSetControlMSG, L801: OBTAIN_CONTROL_SUCCESS

[...]

STATUS/1 @ hpThread, L265: cnt=30

ERRORLOG/1 @ hpThread, L262: Update radius failed
STATUS/1 @ getErrorCodeMessage, L656: hpThread
STATUS/1 @ getCMDIDSetControlMSG, L801: OBTAIN_CONTROL_SUCCESS

ERRORLOG/1 @ sendInterface, L347: ERROR,there is not enough memory

```
**Simulator**
[image]

imkhairulikhwan, 2018-11-27:
Hello Jonathan, 

may I know, did you run the simulator in Linux? I'm running M210 samples in Ubuntu 16.04. It will be really great to have it running in the simulator. From my understanding, only Windows and MacOS have DJI Assistant, to run the simulator.

Thank you!

jonathanmichel, 2018-11-27:
Hello,

I'm using the simulator on Windows. It seems that it is only available on Windows and MacOs.

imkhairulikhwan, 2018-11-27:
Thanks for the clarification. From the hardware setup in https://developer.dji.com/onboard-sdk/documentation/development-workflow/hardware-setup.html, if we were to run Linux samples in onboard computer, we would have to connect the USB-USB to Windows laptop and USB-TTL to the onboard computer itself. 

Which means, only samples without Advanced Sensing?

jonathanmichel, 2018-11-27:
I didn't try advanced sensing for now. It's seems that you can't use advanced sensing and simulation at the same time, take a look [here](https://developer.dji.com/onboard-sdk/documentation/M210-Docs/simulation-checklist.html) and [here](https://developer.dji.com/onboard-sdk/documentation/M210-Docs/oes-checklist.html). You have to turn off simulation to use advanced sensing.

imkhairulikhwan, 2018-11-27:
Sorry, what I was trying to say, is it correct that those are your connection when running the OnBoard SDK? 

Windows Laptop <--(USB USB)-- M210 --(USB TTL)--> Onboard computer 

Just to confirm as I was not able to get it running using this connection.

jonathanmichel, 2018-11-27:
Yes it is correct. If you want more help, you should open a new issue and we could see what is going on and how to fix your problem.

matspetter, 2019-01-09:
Jonathan, did you resolve this issue? I think I have the same problem as you. I can initiate a hotspot just fine and I can update the radius & yaw rate a few times but after a while I always get the error "ERRORLOG/1 @ sendInterface, L347: ERROR,there is not enough memory". I use the blocking calls of updateRadius() and updateYawRate() and independently of how long timeout I set (I tried up to 15 secs) the calls always blocks for the full timeout. So it seems that no ack is coming in!!??

Interestingly enough I dont use the same SDK ver and controller as you (I think). I use an A3, version 1.7.0.5 and the SDK version 3.2 (I think).

jonathanmichel, 2019-01-10:
I don't know which controller is in the Matrice 210 but indeed I have a custom version of the SDK based on 3.7. Currently I resolved this issue by adding two customs functions in the SDK. That's not the best way to solve this but it works. The idea is to send commands that not require ack. 

See [here](https://github.com/jonathanmichel/Onboard-SDK/blob/5643a3e65e468a8f9a9a5b848f2d22569910c50d/osdk-core/api/src/dji_hotpoint.cpp#L270)  for updateYawRate and [here](https://github.com/jonathanmichel/Onboard-SDK/blob/5643a3e65e468a8f9a9a5b848f2d22569910c50d/osdk-core/api/src/dji_hotpoint.cpp#L313) for updateRadius. 

Please note that I mainly use radius update, I had some strange behavior with yaw rate modifications.

matspetter, 2019-01-10:
:) I solved it exactly the same way ... by using type 0 (no acks). I now think this an error in the SDK since when looking here: https://developer.dji.com/onboard-sdk/documentation/protocol-doc/open-protocol.html (search for "Set radius") one can see that this particular command doesn't use ack.

closed 2021-06-24 by dji-kyle