dji-sdk/Onboard-SDK #392: DJI M600 - Subscribtion error

dell640, 2018-12-04:
Hey, I'm unsuccesfull with linux telemetry example on DJI M600.
I have compiled **untouched 3.7 release** [(here)](https://github.com/dji-sdk/Onboard-SDK/releases/tag/3.7.0) on RPi, filled out UserConfig, connected DJI M600 API port, allowed app in assistant, however after succesfull connecting, obtaining SN, Activation it fails with 
**SUBSCRIBER_ILLEGAL_UID**

**M600 has fully updated FW (1.0.1.66 - 3.2.41.13)** which is listed as compatible with 3.7 version.

Any suggestions?

_bash output of ./djiosdk-telemetry-sample_
--------------------
Read App ID
User Configuration read successfully.
STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 115200...
STATUS/1 @ init, L65: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = 0670146254
STATUS/1 @ parseDroneVersionInfo, L708: Hardware = PM820V3PRO
STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.2.41.13
ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!
STATUS/1 @ activate, L1277: version 0x302290D
STATUS/1 @ activate, L1290: Activation successful
STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L869: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.
STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L913: Gimbal not mounted!
| Available commands:                                            |
| [a] Get telemetry data and print                               |
| [b] Select some subscription topics to print                   |
| [c] Get telemetry data and save to file                        |
b

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 0 result: 34.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=8.

STATUS/1 @ getErrorCodeMessage, L656: startPackage
STATUS/1 @ getCMDSetSubscribeMSG, L739: SUBSCRIBER_ILLEGAL_UID

STATUS/1 @ getErrorCodeMessage, L656: subscribeToDataForInteractivePrint
STATUS/1 @ getCMDSetSubscribeMSG, L739: SUBSCRIBER_ILLEGAL_UID

STATUS/1 @ getErrorCodeMessage, L656: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L739: SUBSCRIBER_PACKAGE_DOES_NOT_EXIST

STATUS/1 @ reset, L567: Reset Subscription Successful.
root@Alterum:/Onboard-SDK-3.7.0/build/bin#

kyrie0109, 2018-12-10:
same problem…

dell640, 2018-12-10:
Thanks for confirmation... :(
I'm in touch with dev@dji.com, however not succesfull -- this is their reply:

> Dear costumerThanks for contacting DJI!
> Only A3 and N3 support “TOPIC_RC_FULL_RAW_DATA”. M600 Pro has the different firmware with A3,so this function can be called by M600 Pro.
> Thanks for your understanding and support!
> Best Regards DJI Dev Team 

... so they did not even read the request (has been tested on DJI M600 Pro), and additionally M600 is also listed as supported for 3.7 OSDK.
I will try to ask again...

kyrie0109, 2018-12-10:
> Thanks for confirmation... :(
> I'm in touch with [dev@dji.com](mailto:dev@dji.com), however not succesfull -- this is their reply:
> 
> > Dear costumerThanks for contacting DJI!
> > Only A3 and N3 support “TOPIC_RC_FULL_RAW_DATA”. M600 Pro has the different firmware with A3,so this function can be called by M600 Pro.
> > Thanks for your understanding and support!
> > Best Regards DJI Dev Team
> 
> ... so they did not even read the request (has been tested on DJI M600 Pro), and additionally M600 is also listed as supported for 3.7 OSDK.
> I will try to ask again...

Thank you very much!

dell640, 2018-12-10:
Official DJI response is, that M600 & M600 Pro does not support new subscription topics of 3.7 OSDK
Only A3 & N3 are supported.

....

lbdhr, 2019-01-12:
I have tried the telemetry sample of Linux platform. Once I select b) option of the sample, it only printed `STATUS/1 @ reset, L567: Reset Subscription Successful.`. Then the program exited automatically. I am curious why this happened? It seems no error occurred.

koyahatataku, 2019-01-17:
It seems updating osdk 3.6 to 3.7 with M600Pro doesn't make sense, doesn't it?

zthorson, 2019-01-31:
I spent way too much time digging through this, and here are my conclusions:

M600 support in the 3.7 OSDK is spotty and has a lot of gotchas with the new topics.  In particular, there are a large number of topics that DJI put into the sample code that are not implemented on the M600.  In their documentation, they have the following note on quite a few topics:

Note:
This topic was added in August 2018. Your aircraft may require a FW update to enable this feature. M600 platform support will be added after a firmware update is released.

This reference can be found here:
https://developer.dji.com/onboard-api-reference/group__telem.html

Though, since this OSK was updated in August, and it is now nearly February, I highly doubt that they will ever bring topic support for these topics over to the M600.

For now, if you use the 3.7 OSDK and which to try running some sample telemetry, you can use the patch I have attached to my post.  This removes from the sample application all topics which the M600 does not support.

zthorson, 2019-01-31:
Patch file to run telemetery example.
[m600_telemetry_sample.zip](https://github.com/dji-sdk/Onboard-SDK/files/2818713/m600_telemetry_sample.zip)

dell640, 2019-03-05:
Hey all,
M600 pro got a Firmware update a few days ago, can somebody confirm that topic "TOPIC_RC_FULL_RAW_DATA " is now working on DJI M600 or not? Release notes are not clear...

xelmirage, 2019-04-17:
just done a test，for the topics in the option c in the example, these topics have to be commented to run on M600 Pro

```
TOPIC_RC_WITH_FLAG_DATA
TOPIC_RTK_CONNECT_STATUS
TOPIC_POSITION_VO
```

radjkarl, 2019-06-03:
> Hey all,
> M600 pro got a Firmware update a few days ago, can somebody confirm that topic "TOPIC_RC_FULL_RAW_DATA " is now working on DJI M600 or not? Release notes are not clear...

TOPIC_RC_FULL_RAW_DATA causes SUBSCRIBER_ILLEGAL_UID on my M600pro

kjschill, 2019-06-06:
@dji-dev any updates on when/if this TOPIC will be added to M600 or M600Pro? This functionality is need for reading button states on the controller.

tkazik, 2020-09-08:
I wish I saw this issue earlier.
We had a similar issue with the TOPIC_ESC_DATA and unfortunately just got a reply by DJI that even the newest firmware does not support it (for the interested reader: [here](https://github.com/dji-sdk/Onboard-SDK/issues/627#issuecomment-688607212)). I would be surprised if they released another firmware supporting all the 'new' topics from version 3.7.
Too bad that DJI's documentation is not clearer.
Thx for the patch @zthorson!

closed 2021-06-24 by dji-kyle