dji-sdk/Onboard-SDK #394: How to control the drone by speed via function velocityAndYawRateCtrl?

tong-zha16, 2018-12-12:
Hi,
I'm a user of ONBOARD SDK of linux platform. The stock DJI OSDK 3.7 came with several flight_control_samples.cpp and main.cpp locations such as one in sample>linux>flight-control and another in build>bin>flight-control-sample. In the sample they used positionAndYawCtrl in line 470 of flight_control_samples.cpp to control the drone by position. I'd like to remote control the drone, thus I need to control by speed. I look up to the reference and I found there is a function called velocityAndYawRateCtrl. Are there any samples that the drone is controlled by speed? Or may I know how to modify the flight control sample to control the drone by speed?T he OSDK on DJI documentation does not mention much about these thus i'm really lost as i'm new to SDKs too. Please help me!

I'm running ubuntu and M600pro , samples can run however i need to do more based on them.

Thank you in advance!

ol-bones, 2018-12-13:
The velocity argument takes floats for velocity in metres per second, and should be called ~30 times per second

tong-zha16, 2018-12-13:
> The velocity argument takes floats for velocity in metres per second, and should be called ~30 times per second

Thank you very much! You mean I should put velocityAndYawRateCtrl in a circle and keep calling it? And could you tell me how can I get these information(e.g ~30 times per second)?

ol-bones, 2018-12-13:
Yes.

Documentation doesn't specify this but velocityAndYawRateCtrl resets on the flight controller to avoid flyaway in case of error.

tong-zha16, 2018-12-13:
> Yes.
> 
> Documentation doesn't specify this but velocityAndYawRateCtrl resets on the flight controller to avoid flyaway in case of error.

Thanks! Could I know what do you mean by "velocityAndYawRateCtrl resets on the flight controller"?You mean the drone would reset the velocity to zero at a certain frequency, so we have to keep calling the velocity control function?

ol-bones, 2018-12-13:
yes

tong-zha16, 2018-12-19:
> yes

Thank you very much!I finally manage to move my drone via speed control! And I also get response from DJI, they send me this sample.
[image]

shahardinar, 2019-02-05:
Hi.
I am working now on something very similar.
An onboard drone that will be controlled by speed.
Can I look at your project and learn from it ?
Thank you,
Shahar

closed 2021-06-24 by dji-kyle