dji-sdk/Onboard-SDK #404: I cannot run the sample OSDK app

samanaz, 2019-01-18:
Hi

I am trying to run the sample OSDK app but I face the following error. I have installed everything correctly but I think the activation has a problem. This is the error output:

pi@raspberrypi:/share/dji/Onboard-SDK/build/bin $ sudo ./djiosdk-flightcontrol-sample UserConfig.txt
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyS0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L706: Device Serial No. = removed for security reasons by me

STATUS/1 @ parseDroneVersionInfo, L708: Hardware = A3

STATUS/1 @ parseDroneVersionInfo, L709: Firmware = 3.3.8.47

ERRORLOG/1 @ initVirtualRC, L1067: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1277: version 0x303082F

STATUS/1 @ getErrorCodeMessage, L656: activate
STATUS/1 @ getCommonErrorCodeMessage, L722: ACK_NO_RESPONSE_ERROR

ERRORLOG/1 @ activate, L1302: Failed to activate please retry SET 0x0 ID 0x0 code 0xFFFF

STATUS/1 @ getErrorCodeMessage, L656: initVehicle
STATUS/1 @ getCommonErrorCodeMessage, L722: ACK_NO_RESPONSE_ERROR

ERRORLOG/1 @ sendData, L407: Port closed.
ERRORLOG/1 @ sendData, L411: Open Protocol cmd send failed, send_len: -1 packet_len: 19

STATUS/1 @ getErrorCodeMessage, L656: reset
STATUS/1 @ getCommonErrorCodeMessage, L722: ACK_NO_RESPONSE_ERROR
Vehicle not initialized, exiting.

samanaz, 2019-01-18:
Would you please help me to fix this since I am new to DJI OSDK development.

SChalice, 2019-03-13:
What drone are you using? It seems that the initial communications (Vehicle Initialize) works because you are getting a version number but communications seems to fail when you try to push the Activate button because it says "L722: ACK_NO_RESPONSE_ERROR". 

In Activate() this line of code appear to be failing:

  if (ack.data == OpenProtocolCMD::ErrorCode::ActivationACK::SUCCESS &&
      accountData.encKey)

Which is simply the drone not responding to the message or an encryption problem? 

Does the DJI Go program on the remote say that your connection is good?

closed 2021-06-24 by dji-kyle