dji-sdk/Onboard-SDK #405: OSDK sample applications from Windows

pw24, 2019-01-23:
Has anyone had the requirement to try and incorporate the OSDK from a Windows environment? I know that DJI as released the Windows SDK Beta, but this only supports functionality for Mavic and Phantom range so far... where I need to interface to Matrice range. 

The Windows <-> Linux compiling compatibility is getting me down, where I described the problem on [StackOverflow](https://stackoverflow.com/questions/54332555/dji-osdk-in-windows). Some sample applications through Windows would really be awesome?

lanyusea, 2019-01-24:
The OSDK is a protocol which has no limit about the working environment/machine but just follow the protocol content.

There is a QT implementation of OSDK, which is runnable on Windows platform, see if it helps.
https://github.com/dji-sdk/Onboard-SDK/tree/3.7/sample/Qt

lanyusea, 2019-01-24:
@pw24  maybe read the guide first https://developer.dji.com/onboard-sdk/documentation/quick-start/index.html

pw24, 2019-02-12:
@lanyusea, I have scrolled through the documentation site... not sure if you're referring to something sections I didn't focus on? (which specifically refers to implementation for Windows)

Regarding implementation through Qt... (and this is where I'm fuzzy) No experience with Qt... but need to make the API executable to a .lib / .dll (so that I can encapsulate the API into a C# environment - wherein inhouse company programs are developed)

What I have however been doing in the mean time (as I thought it _better / easier_) was to include the c++ files into a Visual Studio project, to generate the .lib. From this I saw all the linux-dependent classes / code (which isn't all that much actually... just Linux_serial_device class, and one / two other), and then decided to rewrite those classes with exactly the same functions, with with Windows-relative code. Then just exchange the references to the linux classes to Windows classes.

I really think this can work!! I am however very fuzzy with linux-related c++ system classes / code. So if someone can tell me what a few of the functions mean or do, I'll be able to find the Windows-version of that.

Don't know if someone is able to help me... please? I can post my version of the code in a repo, or branch this repo if someone is able to guide me a bit on the Linux side?

[image]

lanyusea, 2019-02-13:
@pw24 

maybe check the QT ones: https://github.com/dji-sdk/Onboard-SDK/tree/d43cf0ab135b015145ab7a2e41b85cbee6522f83/osdk-core/platform/qt/src

I understand what you try and the idea is correct: you need to call the serial read/write API of Windows instead of Linux.

also as @jasperza says, you can refer to the Windows SDK:https://github.com/dji-sdk/Windows-SDK since you're working on Visual Studio. It may be more convenient for your development.

pw24, 2019-02-13:
@lanyusea 

The Windows SDK is not compatible for Matrice range _yet_ (only for Phantom and Mavics... don't know why only that mentioned). 

> ### Supported Products
> The supported DJI products include Mavic 2 Pro/Zoom, Phantom 4 Pro V2.0+ (remote controller with > screen), Mavic Air and Phantom 4 Pro V2.

Yeah, seems like the Qt files are more simplistic / straight forward (Hope that doesn't affect the robustness of the implementation). If that is the case, I think the majority of my Windows serial class is finished.... But will give it a twirl, and see what I end up with, and let know

lanyusea, 2019-02-13:
@pw24 good luck buddy!

sorry I forgot you mentioned it in your original post, my fault

pw24, 2019-02-26:
Hi everyone, I'm still struggling with this issue. 

At some point it tried to load the Qt example in Visual Studio via [Qt Visual studio tools](https://marketplace.visualstudio.com/items?itemName=TheQtCompany.QtVisualStudioTools-19123), (because still need to C# environment) but opening the project produced 300+ errors. So didn't pursue that further at this point.

From the c++ out of Win32 environment I wrote serial comms class, included pthread DLL dependencies into project, and made additional alterations (#if _WIN32 where seemed required). Got it compiled and started testing, but doesn't seem to get any command reply through the connection, with this error:

> Hello Test
>
> STATUS/1 @ DJI::OSDK::Win_serial_device::init, L47: Attempting to open device COM21 with baudrate > 230400...
>
> STATUS/1 @ DJI::OSDK::Win_serial_device::init, L75: ...Serial started successfully.
>
> STATUS/1 @ DJI::OSDK::Win_serial_device::_PortParams, L202: Port parameters successfully set
> STATUS/1 @ DJI::OSDK::Win_serial_device::_PortTimeouts, L129: Port timeout parameters successfully > set
> STATUS/1 @ DJI::OSDK::Win_serial_device::_PortReceive, L102: Port data reception enabled
> STATUS/1 @ DJI::OSDK::Win_serial_device::serialConfig, L92: .. Serial Config done, Successful:
> ERRORLOG/1 @ getDroneVersion, L1378: Drone version not obtained! Please do not proceed.
> Possible reasons:
>        Serial port connection:
>                * SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
>                * Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
>                * TX and RX pins are inverted.
>                * Serial port is occupied by another program.
>               * Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to  replace $USER with your username). Then logout and login again
>
> ERRORLOG/1 @ activate, L1267: Unable to initialize some vehicle components!
> STATUS/1 @ getErrorCodeMessage, L656: Activate
> STATUS/1 @ getCommonErrorCodeMessage, L722: ACK_NO_RESPONSE_ERROR
> 
> STATUS/1 @ DJI::OSDK::Win_serial_device::Close, L321: ...Serial port close successful
> 
> Press any key to continue . . .

Also validated that _something_ is written to the COM port (through COM emulator and hyperterminal), and switched Rx / Tx connections, but no reply is received (also tried changing baud through DJI Assistant 2, not sure but doesn't seem to be working?). 

I don't know anymore... someone maybe able to please help and tap in?

lanyusea, 2019-02-26:
> also tried changing baud through DJI Assistant 2, not sure but doesn't seem to be working?

the baudrate should be matched between the drone setting and your program param

pw24, 2019-02-26:
Jip, the default baud to platform is 230400 as far as I'm aware. So kept to this baud rate in my application. 

But wanted to test a few different bauds, because of Win's serial CBD doesn't have that baud enum, but after reading on stackOverflow, it was mentioned that it doesn't matter. You can set the int baudrate value directly, and don't need to use CBR:

```
DCB dcbSerialParams = { 0 };
dcbSerialParams.DCBlength = sizeof(dcbSerialParams);
deviceStatus = GetCommState(m_serialHandle, &dcbSerialParams);	//get current settings of port

//configure baudrate
dcbSerialParams.BaudRate = baudrate;
/*int st_baud[] = { CBR_4800,  CBR_9600, CBR_19200, CBR_38400,
				CBR_57600, CBR_115200, CBR_256000 };
int std_rate[] = { 4800,   9600,   19200,   38400,   57600,  115200, 256000 };
for (int i = 0; i < sizeof(st_baud); i++)
	if (std_rate[i] == baudrate)
	{
		dcbSerialParams.BaudRate = st_baud[i];
		break;
	}*/
```

But to use different bauds to the RPA, I'm under the impression the RPA's Comm port baud must be changed through DJI assistant as well. But everytime I change it and restart the program, it again shows 230400 (so doesn't look as if changed successfully). So not possible to test _int baudrate value_ vs. _CBR baudrate value_ through the program, if the RPA's gonna be stuck on 230400.

Don't know if anyone has a M210 and willing to test my program? So that I can determine if my problem is hardware / software based? I've published my Windows based code on a **[separate repo](https://github.com/pw24/DJI-OSDK-Win32_console)** which I can integrate into this project if / when eventually works?.

lanyusea, 2019-02-28:
@pw24 

just noticed this issue, may be some hardware issue:
https://github.com/dji-sdk/Onboard-SDK-ROS/issues/215#issuecomment-467865695

SChalice, 2019-03-13:
I am running the OSDK under Windows 7. I used the QT sample and converted it to VC using the VS QT tools. I am looking for a tool to convert the rest of the QT portion of the project to pure MFC and then my SDK will be much superior to the Windows SDK. I also have wireless communications going between my Windows OSDK app and my Android MSDK app.

It is totally possible.

Cheers.

pw24, 2019-03-14:
Hi @SChalice , thanks for the reply. Jip, looks like we went through exactly the same learning curve. 

The porting from linux version didn't go as expected. Wrote my own serial class and included PThreads for Win32 + minor additional changes throughout, but the connection still wasn't successful. Got the idea that the problem could be in the _#ifdef_ sections, but wasn't very keen to start rewriting stuff in the OSDK (can't justify the time required to traverse through the whole OSDK).

Eventually went through QT VS tools as well. (our comms is with M210). At the moment have a c++ server with OSDK running on the RPA single board computer, but actually preferred to export c++ code to DLL for usage in C# environment (to integrate with rest of our development, integration with hardware, sensors etc.). 

And... for some reason encapsulated OSDK functions execute through c++ application, but when executing those **EXACT** functions through DLL from c# environment, it's unsuccessful?? (boggles my mind) 

Anyway, all my progress is repo-ed [here](https://github.com/pw24/Qt_DJI_OSDK_DLL/tree/Server2.0) (just not documented) until it's ready to contribute onto this repo. @SChalice , don't know if you would be interested in communicating a few of the aspects seeing as though we're doing the same thing? 

My next step is to get CLR DLL running for the OSDK from c# (for debugging purposes to fix whatever the problem is for interfacing from c# environment). But the Qt dependency is buggering me around

closed 2021-06-24 by dji-kyle