dji-sdk/Onboard-SDK #410: M210 F/W V01.02.0000 OpenProtocol::VerifyHead() error

jaesik-hwang, 2019-02-12:
- M210, F/W V01.02.0000
 - OnBoardSDK 3.7 with STM32F.

At OpenProtocol::verifyHead(), p_head->reserved1 is not zero! (is 0x313058)
When I send a get drone version packet with following function tree.
 v->functionalSetUp();
      initVersion()
            getDroneVersion();

I have received some bytes from M210 and it's sof, version, length and reserved0 is correct. but reserved1 is not zero. So the received packet is ignored.

I have printed all received bytes at the receive interrupt. That is...
AA 41 00 00 FF 58 **30 31 53** 76 20 41 34 9A

As received bates saying, received packet's reserved1 is 0x313058.
I think M210 returns wrong packet or header definition is changed.
As OnBoardSDK 3.7 > OpenProtocol::verifyHead() is defined like following codes reserved1 must be zero.

```
if ((p_head->sof == OpenProtocol::SOF) && (p_head->version == 0) &&
      (p_head->length < OpenProtocol::MAX_RECV_LEN) &&
      (p_head->reserved0 == 0) && (p_head->reserved1 == 0) &&
      (crcHeadCheck((uint8_t*)p_head, sizeof(OpenHeader)) == 0))
```

lanyusea, 2019-02-12:
looks weird.
could you try to downgrade the firmware version to 1.1.0913 as https://github.com/dji-sdk/Onboard-SDK#firmware-compatibility says to try again?

jaesik-hwang, 2019-02-14:
How can I downgrade it with 1.1.0913? 
V0.1.02.0000 is oldest version in my DJI Assistant 2 for Matrice.
I got M210 two weeks ago.

[image]

Be sides, Baud Rate and sampling rates of telemetry values are not changed in SDK tap of DJI assistant for Matrice Windows App.
[image]

I think those problems are related communication problem on newest firmware.

For the record, I did
1. firmware upgrade and downgrade with legacy DJI Assistant 2(V1.2.4)
   because I didn't know DJI Assistant 2 for Matirce is existed at this time. 
   a)  V01.02.0000 --> V01.02.0301
   b) V01.02.0301 --> V01.02.0000

2. firmware upgrade and downgrade with DJI Assistant 2 for Matrice(V2.0.4)
   a)  V01.02.0000 --> V01.02.0301
   b) V01.02.0301 --> V01.02.0000

3. RESTORE FACTORY DEFAULTS with DJI Assistant 2 for Matrice(V2.0.4)

And between firmware up/down grades I have tried API connection and changing "Baud Rate" and failed all.

jaesik-hwang, 2019-02-18:
Increasing cpu clock speed from 26MHz to 206MHz solve this problem.
(I use custom board instead of STM32F4 evaluation board and use own BSP(board support package).)

All received packets are processed at interrupt handler and there is no FIFO buffer. So some receiving bytes are ignored when cpu clock speed is not enough.

By the way, SDK configuration fail at DJI Assistant 2 for Matrice problem is still existed.

closed 2021-06-24 by dji-kyle