dji-sdk/Onboard-SDK #418: How to continuously control aircraft movement？

fuying-wang, 2019-03-06:
Hi, 
I am working on a drone controlled project using DJI Onboard SDK. In my mission, the target point of the aircraft is changing at a certain frequency such as 50Hz. So I need to move the aircraft to the target point  in each loop. 
But I met a problem that when I continued to send `vehicle->control->velocityAndYawRateCtrl`
instructions to the aircraft, the aircraft moved at the beginning, but after some loops, the speed is reduced to approximately equal to zero. Even if I reduce the command frequency I still meet this problem. I am a newcomer to drone control so I am also not sure if using velocity to control drone is appropriate in my task. Or may be I need using the`positionAndYawCtrl` function?
So could anyone help me with this problem? Thank you very much!

lanyusea, 2019-03-06:
what did you send and how.

could you paste your code here

fuying-wang, 2019-03-07:
Hi, 

Sorry for my late reply.

In my task,  I have created two threads. One thread is used to update the command (velocity) information, and another thread keeps sending instructions to the aircraft. Here is the flight control thread:

```c++
int timeoutInMilSec = 60 * 1000;
int elapsedTimeInMs = 0;
int cycleTimeInMs = 20;

while(elapsedTimeInMs < timeoutInMilSec)
{
    myVehicle->control->velocityAndYawRateCtrl(
        moveVelocity[0], moveVelocity[1], moveVelocity[2], 0);
    usleep(cycleTimeInMs*1000);
    elapsedTimeInMs += cycleTimeInMs;
}
```

Thank you very much!

lanyusea, 2019-03-07:
looks good,

what value did you send?

fuying-wang, 2019-03-07:
The `moveVelocity` I sent is between -10 and 10. 
In detail, when the distance between vehicle and target point is too large, the velocity will be set to 10, when the distance is smaller, the velocity also will be smaller.

lanyusea, 2019-03-07:
will it still happen if you keep sending 10?

If so I guess it may reach the distance limit, which you can check in DJI GO

fuying-wang, 2019-03-07:
Yes, it exists. 
Thank you very much!  I will check it in the next experiment.

SChalice, 2019-03-13:
I am confused.

Is it hitting the arbitrary distance limit that you can manually remove? Do you need to be logged in to DJI servers on the remote? Is it a country restriction? Or is this a wireless communication distance thing?

fuying-wang, 2019-03-14:
Hi, Thank you for your reply. 
Sorry for my carelessness and it's a problem of not connecting remote control correctly.

closed 2019-03-22 by fuying-wang