dji-sdk/Onboard-SDK #419: Obtaining control in DJI Matrice 600 pro without RC

vAnArhist, 2019-03-07:
Hello! Checking the Onboard-SDK API (test sample) with STM32F4Discovery board. In simulator mode seems to be ok, but how can I obtain control on real drone, if Virtual RC has been deprecated and is unavailable for A3 FW > 1.6.0.0.
Now it doesn`t working on latest firmware on A3 (1.7.x.x)
We are using A3 Flight controller and main goal is to control drone without RC. Is it possible to downgrade the FW of A3?

[image]

lanyusea, 2019-03-07:
@vAnArhist set the Failsafe option to Hover in DJI Assistant 2

vAnArhist, 2019-03-07:
@lanyusea It is set. Remote controller is on and in F mode.

SChalice, 2019-06-25:
I am using the QT OSDK with an onboard nano pc running Windows connected to the A3 with a ttl cable.
I can obtain control without an RC. :)

closed 2021-06-24 by dji-kyle