dji-sdk/Onboard-SDK #423: issue on Advanced Sensing - Stereo Depth Perception sample

mohand150, 2019-03-17:
hello 
DJI Developer and Developer Advanced Detection -> Building the OSDK and Samples
[https://developer.dji.com/onboard-sdk/documentation/sample-doc/sample-setup.html](url)
I want to test the example of Advanced Sensing samples - Stereo Depth Perception on Jetson TX2 ( opencv 3.4 + opencv_contrib 3.4 + cuda 9.0)
I start by downloading the project
mkdir build
cd build
cmake .. -DADVANCED_SENSING = ON
make
the build stops with an error in using a variable LIBUSB_1_INCLUDE_DIR

then I did the same process on my pc with **ubunut 16.04** (see the file error) + opencv3.4 + opencv_contrib 3.4
I get the same error .... 
```

mohand@ThinkPad:~/Onboard-SDK-3.7/build$ cmake .. -DADVANCED_SENSING=ON
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- External library - DJI Advanced Sensing for stereo cameras
-- Found FFmpeg  installed in the system.
--  - Includes: 
--  - Libraries: /usr/lib/x86_64-linux-gnu/libavcodec.so;/usr/lib/x86_64-linux-gnu/libavformat.so;/usr/lib/x86_64-linux-gnu/libavutil.so;/usr/lib/x86_64-linux-gnu/libswscale.so;/usr/lib/x86_64-linux-gnu/libvorbis.so;/usr/lib/x86_64-linux-gnu/libvorbisenc.so;/usr/lib/x86_64-linux-gnu/libtheora.so;/usr/lib/x86_64-linux-gnu/libz.so
-- AdvancedSensing sample is enabled

stereo-vision-single-thread-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/include;/usr/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab

stereo-vision-multi-thread-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/include;/usr/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab

camera-stream-poll-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/include;/usr/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab

camera-stream-callback-sample...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/include;/usr/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab

stereo-vision-depth-perception-sample...
-- Found OpenCV 3.3.1 installed in the system, will use it for depth perception sample
--  - Includes: /usr/include;/usr/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab
-- Did not find ximgproc in OpenCV
-- Did not find viz3d in OpenCV
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
[... 39 lines cut ...]
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/camera_stream_callback_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/camera_stream_callback_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/camera_stream_callback_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/camera_stream_callback_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/stereo_vision_depth_perception_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/stereo_vision_depth_perception_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/stereo_vision_depth_perception_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/stereo_vision_depth_perception_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/stereo_vision_depth_perception_sample
   used as include directory in directory /home/mohand/Onboard-SDK-3.7/sample/linux/advanced-sensing/stereo_vision_depth_perception_sample
LIBUSB_1_LIBRARY
    linked by target "djiosdk-core" in directory /home/mohand/Onboard-SDK-3.7/osdk-core

-- Configuring incomplete, errors occurred!
See also "/home/mohand/Onboard-SDK-3.7/build/CMakeFiles/CMakeOutput.log".
See also "/home/mohand/Onboard-SDK-3.7/build/CMakeFiles/CMakeError.log".

```
[CMakeError.log](https://github.com/dji-sdk/Onboard-SDK/files/2975772/CMakeError.log)
[CMakeOutput.log](https://github.com/dji-sdk/Onboard-SDK/files/2975773/CMakeOutput.log)

**
I guess I still miss some steps, maybe?

Thank you in advance!**

lanyusea, 2019-03-18:
@mohand150  did you install the libusb?
if so, try -DLIBUSB_1_INCLUDE_DIR=... to specify the libusb's header file directory

mohand150, 2019-03-18:
@lanyusea thank you for your help

closed 2019-03-18 by mohand150

zongoalbert, 2020-07-16:
@lanyusea  @mohand150  same issue please need your helps , thanks

kpsuhair, 2021-09-07:
Please check libusb & libusb-dev is installed. If not installed ,install it by using 

sudo apt-get install libusb-1.0-0-dev