dji-sdk/Onboard-SDK #424: M210 UART talk to TX2 UART directly

wt70707, 2019-03-18:
I have M210 that can talk to NVidia TX2 through USB FTDI (UART) cable, but cannot talk directly through the UART on NVidia TX2. The baud rate is set to match on DJI Assistant 2 and on TX2. What is the cause, that FTDI cable works but direct UART doesn't work?

lanyusea, 2019-03-18:
@wt70707 can you find any input data from the UART port by using command `cat` in TX2?

wt70707, 2019-03-18:
@lanyusea Yes, I can get data in TX2. By cat, or stty or putty, the data is a mess. With "od -x", it can be seen different frames of data. I got errors of "drone activation error" and "Vehicle initialization failed" when I run dji_sdk ROS launch node ($roslaunch dji_sdk sdk_vm.launch).

lanyusea, 2019-03-18:
looks strange if the FTDI works well.

I guess maybe the TX2 serial is not correctly configured. I suggest to also ask this question in https://devtalk.nvidia.com

wt70707, 2019-03-18:
Yes it is strange. I use osdk-ros, the difference of ftdi cable and user is
the port name, which can be changed in launch file. I also tried to use the
highest baud rate (1000000) with the ftdi cable, but it does work also. Use
stty, it shows the baud rate is 0.

wt70707, 2019-03-18:
I had talked with someone on devtalk.nvidia.com. From the Nvidia TRM
Errata, there is one for UARTs:

"UART-1 does not meet +/- 4% Baud Tolerance. Tegra RX has limitation in one
stop bit mode configuration that its baud rate needs to be always higher
than that of transmitter. Otherwise, the received data will have
corruption/invalid framing errors. Use 2-stop bits configuration wherever
possible. "

Advice goes to using two stop bits, not sure if it will work and on "DJI
Assistant 2 For Matrice", I can only change the baud rate.

lanyusea, 2019-03-19:
@wt70707  so does it help after you increase the baudrate?

mhoejgaard, 2019-03-21:
@wt70707 We have experienced a similar issue and have not been able to solve it. We fell back to use FTDI.

wt70707, 2019-03-21:
@lanyusea It works with baudrate of @230400 and @115200, but not work @921600 and @38400.

wt70707, 2019-03-21:
@mhoejgaard This is what we also did.

snakehaihai, 2019-04-04:
For TX2 or manifold 2, when you connect it to M210, there are a few tricks. 
1. Get correct FDTI cable. most fundamental error. 
2. Not all TX2 /MF 2 firmware works. DJI once passed beta version of MF2 to us with the latest firmware they didn`t even tested. Turns out not working and need to flash back to older version manifold  2 firmware. 
3. USB Permission. standard udev rule

closed 2021-06-24 by dji-kyle