dji-sdk/Onboard-SDK #430: How can I enable API control without the DJI Assistant 2 app?

evagorasmakridis, 2019-04-10:
I work on a Linux Ubuntu 16.04 machine and I would like to enable the API control without the DJI Assistant 2 application as it is not provided for Linux machines.
Is there a way to enable the API control without the DJI assistant?

lanyusea, 2019-04-11:
as I remember you only need to do this once and it will keep open until next firmware update.

so maybe borrow a windows laptop or install a virtual machine or try wine?

evagorasmakridis, 2019-04-11:
Thank you lanyusea.
I tried wine, I managed to installed the DJI Assistant 2 but then I can not run the .exe.
But if I can do it once with another machine then it is fine. I will try that.
Thank you again!

lanyusea, 2019-04-11:
I will try if I can arrange a Linux version of DJI Assistant 2.

maybe not full version, but a lite one for SDK users

closed 2021-06-24 by dji-kyle