dji-sdk/Onboard-SDK #431: How can I control aircraft using acceleration information?

fuying-wang, 2019-04-15:
Hi, I am doing a project about drone formulation on DJI Onboard SDK.    

According to the DJI Onboard SDK document, we can control the aircraft using position command like `vehicle -> control -> positionAndYawCtrl` function or using velocity command like `vehicle -> control ->  velocityAndYawRateCtrl` function.   

Although I noticed that there are acceleration values in the simulator of DJI assistant 2, I didn't find a flight control function using acceleration command. So how can I control the aircraft using acceleration command?  

Your reply will be highly appreciated.

lanyusea, 2019-04-15:
there is no pure acceleration control API, but you can determine the feedforward acceleration when doing velocity control.

check: https://github.com/dji-sdk/Onboard-SDK/blob/25dd11346c87c6b0fc520b5a9711c9bb31e96609/osdk-core/api/src/dji_control.cpp#L340

fuying-wang, 2019-04-16:
Thank you very much for your reply.

I am still confused about how the feedback acceleration affects velocity when I use the function `vehicle -> control -> velocityAndYawRateCtrl`. May you share me some example codes to learn the usage of the feedback acceleration?

lanyusea, 2019-04-16:
it cannot if you use `velocityAndYawRateCtrl` API, which doesn't support the acceleration feedforward

fuying-wang, 2019-04-16:
I am sorry for my unclear expression. I mean I can modify this API and use `AdvancedCtrlData` as the parameter instead of `CtrlData`. So I am wondering how the feedback acceleration parameters in the `AdvancedCtrlData` affects the velocity?  Thank you very much.

lanyusea, 2019-04-16:
theoretically, It determines what acceleration should be when the drone achieves the velocity you set. i.e. the derivative help you draw the velocity curve.

for example if you wanna do fly a circular path, the velocity control way would be:
```c
vel_x = vel* sin theta
vel_y = vel* cos theta
```
but it cannot be a circle without accelerate limitation

by add accelerate limit:
```c
vel_x = vel* sin theta
vel_y = vel* cos theta
acc_x = acc*cos theta
acc_y = acc* sin theat
```

engineerly in real use, the acceleration will be effective slower than the velocity setting since it influence the drone attitude directly.

fuying-wang, 2019-04-16:
Thank you! That's very helpful!

fuying-wang, 2019-04-22:
Hi, may you tell me the limit of feedback acceleration? I am not sure how to set the specific value of acceleration and I didn't find such information in the documentation. Thanks a lot.

lanyusea, 2019-04-22:
should be no limit, but limited by the physical condition that drone's roll/pich angel should not be larger than 35 degree

fuying-wang, 2019-04-22:
OK. Thank you!

closed 2019-04-29 by lanyusea