dji-sdk/Onboard-SDK #433: Getting video from onboard to remote computer

pavankumarbn, 2019-04-22:
Hi, I am trying to get zenmuse x3 mounted on M100 video feed to a remote computer through SSH from onboard computer. I tried to add remote computer IP to sshd_conf file but It is giving "can't open display" issue. Anybody tried to do the same? Please give suggestions to achieve this objective.
Thank you,
Pavan

lanyusea, 2019-04-22:
seems not a sdk related issue

does this help?
https://superuser.com/questions/310197/how-do-i-fix-a-cannot-open-display-error-when-opening-an-x-program-after-sshi

pavankumarbn, 2019-04-22:
@lanyusea Thank you for the reply. I will go through it.

pavankumarbn, 2019-04-26:
Thank you all for your help! Finally, I found an aletrnate solution to achieve the objective using RTMProtocal.

closed 2019-04-26 by pavankumarbn

DimaNyumovich, 2020-02-25:
Hi
What alternative solution did you find? Can you send me link ? I'd like to make  something like this.

pavankumarbn, 2020-02-26:
> Hi
> What alternative solution did you find? Can you send me link ? I'd like to make something like this.

Hello,
Please find the link (https://github.com/pavankumarbn/UAVVideoFeedAccess). 
best wishes.