dji-sdk/Onboard-SDK #435: How to use this velocityAndYawRateCtrl() function？OSDK of linux platform.

huanghuangyinpeipei, 2019-04-28:
i'm new to OSDKs.  the OSDK on DJI documentation does not mention much about this "velocityAndYawRateCtrl()" function.
   I am trying to call like this.But it didn't work.
  thx!
____________________________________________________________
  case 'b':
      monitoredTakeoff(vehicle);
      moveByPositionOffset(vehicle, 0, 6, 5, 0);
      moveByPositionOffset(vehicle, 6, 0, 5, 0);
      for(int i = 0; i < 10000; i++)
      {
        vehicle->control->velocityAndYawRateCtrl(0, 0, 0.5,0);
        usleep(100);
      }
____________________________________________________________

lanyusea, 2019-04-28:
did you obtain control first?

huanghuangyinpeipei, 2019-04-28:
> did you obtain control first?
yes.
I had obtain control authority.  I am founded on the basis of  flight-control sample(OSDK3.7).
Error message is:
"Port closed."
"Open protocol cmd send failed,send_len:22"

huanghuangyinpeipei, 2019-04-28:
> did you obtain control first?
yes.
i had obtain control authority. and i am founded on the basis of flight-control sample(OSDK3.7). 
do you have a demo about  velocityAndYawRateCtrl() funciton?
thx.

huanghuangyinpeipei, 2019-04-28:
@lanyusea 

yes.
i had obtain control authority. and i am founded on the basis of flight-control sample(OSDK3.7).
do you have a demo about velocityAndYawRateCtrl() funciton?

error message:
ERRORLOG/1 @ sendData, L407:port closed.
ERRORLOG/1 @ sendData, L411:Open Protocol cmd send failed, send_len:35

thx.

lanyusea, 2019-04-29:
@huanghuangyinpeipei 

can you receive the drone status correct?
It says there is some connection issue, which I assume should be hardware realated

huanghuangyinpeipei, 2019-04-29:
@lanyusea 
I don't know which drone status do you say.? 
it is Statue of Telemetry? If so i can get the drone status.
and I am checked the hardware wiring.it is no problem to running moveByPositionOffset().i think it is not hardware faulty.

huanghuangyinpeipei, 2019-04-29:
@lanyusea 
hardware device：raspberry pi + A3, USB-TTL.

lanyusea, 2019-04-29:
then the error message may not be related to this issue.

the usage looks good, maybe try "50Hz in 10 seconds" instead of "10000Hz for a second" and see if it works.

for sample code: https://github.com/search?q=velocityAndYawRateCtrl&type=Code

huanghuangyinpeipei, 2019-04-29:
@lanyusea 
thx!  Let me have a try.
Another problem is: the drone flew to the target point, and i use the velocityAndYawRateCtrl + RTK to control the drone landing. How far is the horizontal distance between the two points(actual landed position, target point )?

lanyusea, 2019-04-29:
@huanghuangyinpeipei 

theoretically should be zero if you set `0` in x and y speed.

but if you want to make drone land, you can call the landing API directly: 
https://github.com/dji-sdk/Onboard-SDK/blob/25dd11346c87c6b0fc520b5a9711c9bb31e96609/osdk-core/api/src/dji_control.cpp#L298

closed 2021-06-24 by dji-kyle