dji-sdk/Onboard-SDK #437: dji_telemetry_doc.hpp is outdated

xelmirage, 2019-05-22:
The SDK code is updated to 3.8, but the dji_telemetry_doc.hpp is still 3.7. 

As far as I know, there are many differences between 3.8 and 3.7 regarding to the unit and definitions of the topic values.

hummelslei, 2019-05-28:
Which topics do you specifically refer to? There aren't many differences between 3.8 and 3.7.

xelmirage, 2019-05-28:
As formerly confirmed with dev@dji.com, the unit of TOPIC_ACCELERATION_RAW is changed from m/s2 to g. and this caused a serious misbehave in our program. 
Now we are having a great trust issue with the document. Since we confirmed there is something changed in the sdk-core, the 3.7 dji_telemetry_doc.hpp is now totally not reliable to us. Even there aren't many differences between 3.8 and 3.7 like you said, we don't know what changed and how they changed, because the dji_telemetry_doc.hpp is not changed from 3.7 to 3.8.

closed 2021-06-24 by dji-kyle