dji-sdk/Onboard-SDK #438: Are all the samples compatible with DJI M210RTK V2? (Onboard SDK 3.8)

alejodosr, 2019-05-22:
Hi,

I am testing all the samples (with to baudrates 921600 and 230400) and I have followed all the tutorials related to DJI M210 RTK V2 but I am not able to successfully run any sample.. any help?

[image]

Thank you in advance.

KhristopherK, 2019-05-24:
We are having the same error messages over here. Have someone being able to successfully run samples on the 210 v2?

alejodosr, 2019-05-24:
I have been able to run the samples. Be aware of the proper serial connection for M210RTK v2 (different from v1), here the [user manual](https://dl.djicdn.com/downloads/m200_v2/20190419/M210_V2_M210_RTK_V2_User_Manual_v1.2_EN.pdf). Also, this checklist has to be met:
- For the first time, register at DJI website as a developer, create an app and get the `app id` and the `enc key`.
- For the first time, with your DJI account, activate the M210RTK V2 with DJI Pilot app.
- In DJI Assistant 2 the `baudrate` should be configured to 921600.
- In DJI Assistant 2, in the SDK tab, activate the Onboard SDK API.
- In the DJI Pilot app, go to setting and enable OSDK I/O communication.
- Give permission to the serial port `sudo chmod 777 /dev/ttyUSB0` (check if exists).
- Configure User_Config.txt to include your `app id`, your `enc key` and the correct `baudrate`.
- Be sure the RC is connected (no DJI app required) while running the simulation.

Hope it helps.

closed 2021-06-24 by dji-kyle