dji-sdk/Onboard-SDK #443: Control::flightCtrl(AdvancedCtrlData data) - meaning of avoid_atti_limit_enable/bit 4

matspetter, 2019-06-11:
I have a question regarding the flight control type AdvancedCtrlData: 
https://developer.dji.com/onboard-api-reference/classDJI_1_1OSDK_1_1Control.html#a6bc141f314b7ac82e93f56660bfa3681

In flightCtrl(AdvancedCtrlData) there are some bits/info that I don’t fully understand:

1) what does “avoid_atti_limit_enable”/bit 4 mean? Or maybe I should ask, what does “atti limit” mean? 
2) normally we use flightCtrl(CtrlData data), is “atti limit” enabled when using flightCtrl(CtrlData data) or ?

dji-dev, 2019-06-11:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/24860'>Zendesk ticket #24860</a>:<br><br>Dear Customer,<br><br>Thank you for contacting DJI.<br><br>Sorry that due to the new changes in the this API interface, the problem may need to be turned into SDK requirements, which will be further verified and tested by relevant engineers, so need to ask for more information from you<br>1. The specific purpose for which you use the API interface<br>2. What application to develop, or project description<br>3. A brief introduction of your company<br>4. The DJI product used<br><br>Thank you for your understanding and support, hope you a delighted life.<br><br>Best Regards,<br><br>DJI SDK Support Team

matspetter, 2019-06-11:
Hmm I don't understand at all! I just want to know the meaning of the "avoid_atti_limit_enable" bit. I **don't** want to introduce any new functionality!

dji-dev, 2019-06-13:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/24860'>Zendesk ticket #24860</a>:<br><br>Dear Customer,<br><br>Thank you for contacting DJI.<br><br>Sorry,  "avoid_atti_limit_enable" bit means to remove the attitude limit (tilt Angle) during attitude control mode, but the API is currently unavailable because the protocol is updated.<br>If you have requirements and want to use the API interface, we need to know more about the purpose of the interface and match your situation.So consider providing information if you need it.<br><br>Thank you for your understanding and support, hope you a delighted life.<br><br>Best Regards,<br><br>DJI SDK Support Team

closed 2021-06-24 by dji-kyle