dji-sdk/Onboard-SDK #445: UNKNOWN_ACK_ERROR_CODE while trying to run the Onboard SDK flightcontrol-sample example.

AbhishekGS, 2019-06-25:
Details of the issue:

1) Describe the issue - include any relevant context and the end goal:

Is an RC transmitter and receiver is required to run the flightcontrol-sample demo code?

I want to get started with DJI Onboard SDK. I followed the documentation and executed the flightcontrol-sample code. I have the DJI Assistant 2 and the simulator running. When I execute the code I get the below errors:

`ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!
`
`STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE
`
When I run the mission-sample example I do not get any errors, but I do not see the drone flying in the simulator as well.

However, I can run the sample codes 'telemetry' and 'logging' without any issue. It looks to me that I am unable to invoke the DJI controller.

Please note that I DO NOT have the RC controller.

2) SDK and SDK version: 
DJI Onboard SDK 3.8.1

3) Aircraft/Products: 
DJI N3

4) Code Sample:
Using the sample code: flightcontrol-sample

5) Steps to reproduce: 
1. Connect the Raspberry Pi 3 and DJI N3 using the UART cable appropriately.
2. Power on both the devices.
3. Start the DJI Assistant 2 software on desktop and start the simulator.
4. Log into raspberry pi and execute the command: ./djiosdk-flightcontrol-sample UserConfig.txt.
5. Observe the below errors:

`ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!
`
```
STATUS/1 @ getErrorCodeMessage, L656: Ùr}P]Ùr}Ç
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE
```

Note that I do not have the RC transmitter and receiver connected.

6) Device and OS (ie Samsung Galaxy, iOS, etc):
Raspberry Pi 3 B+

7) Provide any relevant logs and error messages (include screenshots if possible):

Please refer to the output below.

"
```
Read App ID
User Configuration read successfully.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyS0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0DBDF5E0020175

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = N3

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.3.8.39

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1313: version 0x3030827

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L934: Gimbal not mounted!
| Available commands: |
| [a] Monitored Takeoff + Landing |
| [b] Monitored Takeoff + Position Control + Landing |
b

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=10, nTopics=2.

STATUS/1 @ getErrorCodeMessage, L656: ®Ô+X¢]®Ô]
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

STATUS/1 @ verify, L244: Verify subscription successful.
ERRORLOG/1 @ initPackageFromTopicList, L167: package [0] is being occupied.

STATUS/1 @ verify, L244: Verify subscription successful.
ERRORLOG/1 @ initPackageFromTopicList, L167: package [0] is being occupied.

STATUS/1 @ verify, L244: Verify subscription successful.
ERRORLOG/1 @ initPackageFromTopicList, L167: package [0] is being occupied.

STATUS/1 @ verify, L244: Verify subscription successful.
ERRORLOG/1 @ initPackageFromTopicList, L167: package [0] is being occupied.

STATUS/1 @ reset, L567: Reset Subscription Successful.
```
"

dji-dev, 2019-06-25:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/25378'>Zendesk ticket #25378</a>:<br><br>Dear Customer,<br><br>Thank you for contacting DJI.<br><br>Which version of DJI Assistant 2 are you using? you can try this version for simulation testing, https://www.dji.com/downloads/softwares/assistant-dji-2<br>And it can works flightcontrol-sample with DJI Assistant 2 simulator without RC remote control.<br><br>Thank you for your understanding and support, hope you a delighted life.<br><br>Best Regards,<br><br>DJI SDK Support Team

ashibhuvan, 2019-06-25:
I will try later today or tommorrow without the RC controller and see if I get similar answer. I am also using RaspberryPi and have no problem.

AbhishekGS, 2019-06-25:
@dji-dev I use assistant-dji-2-for-autopilot, as it supports DJI N3. I tried your suggestion of using https://www.dji.com/downloads/softwares/assistant-dji-2 with N3. However, the software does not recognize N3 at all. It looks only for DJI inspire.

@ashibhuvan Please try running the Onboard-SDK Linux flight-control sample without using the RC and let me know if you can see the quadcopter flying in the DJI simulator, without the error I mentioned. Thank you in advance.

dji-dev, 2019-06-26:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/25378'>Zendesk ticket #25378</a>:<br><br>Dear Customer,<br><br>Thank you for contacting DJI.<br><br>According to our tests, both DJI Assistant2 and For Autopilot verison can correctly recognize N3, and both can use the simulator to run the DJI OSDK flight-control sample.<br>Run this sample don't need the remote control, but because you if there are any problems or flight simulator was used for the first time, suggest connection LIGHTBRIDGE first, using a RC connection phone (running DJI GO APP), to check whether there is an error information on the APP, and confirmed that the vehicle can be controlled by remote control for simulated flight in DJI Assistant2, then test OSDK - sample.<br><br>hank you for your understanding and support, hope you a delighted life.<br><br>Best Regards,<br>DJI SDK Support Team

AbhishekGS, 2019-06-26:
@dji-dev 

I tried again using the DJI Assistant 2 and DJI Assistant 2-for the Autopilot. I still get the same error messages:

```
ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform! 

STATUS/1 @ getErrorCodeMessage, L656: ®Ô+X¢]®Ô] 
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE 

```

I am using DJI products for the first time. I do NOT have DJI Lightbridge as well. 

Is it necessary to buy a DJI Lightbridge for the first time? If yes, Can I buy an SBUS supported RC transmitter/receiver instead of DJI Lightbridge? For example, Futaba Transmitter and receiver that supports SBUS.

Why is the DJI Go App required?

Do you have documented steps on how to use the DJI Go App with Onboard-SDK for the first time?

Please advice on the next steps.

Thanks,
Abhishek

dji-dev, 2019-06-27:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/25378'>Zendesk ticket #25378</a>:<br><br>Dear customer,<br><br>Thank you for contacting DJI.<br><br>1.We haven't tested third-party remotes, but we recommend using DJI LIGHTBRIDGRE 2 https://www.dji.com/lightbridge-2?site=brandsite&from=insite_search<br><br>2.DJI GO APP is to help you use N3 products and troubleshoot hardware and other problems. It is not necessarily related to OSDK. You can check the user manual of N3<br>https://dl.djicdn.com/downloads/N3/20170825/N3_User_Manual_En_v1.4.pdf<br><br>Have you tested that running flightcontrol-sample in x86-ubutnu works? Instead of using raspberry pi<br><br>Best Regards,<br>DJI SDK Support Team

Hatyman, 2019-08-07:
I have the same problem(
I use virtualBox with x64 Linux Mint and at the same time win10 with DJI Assistant 2. Connection is well, but it doesn't work.
`Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0DBDF5E0020172

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = N3

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.3.8.47

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1313: version 0x303082F

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L934: Gimbal not mounted!
| Available commands: |
| [a] Monitored Takeoff + Landing |
| [b] Monitored Takeoff + Position Control + Landing |
a

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=10, nTopics=2.

STATUS/1 @ getErrorCodeMessage, L656: F l~
STATUS/1 @ getCommonErrorCodeMessage, L726: UNKNOWN_ACK_ERROR_CODE

STATUS/1 @ verify, L244: Verify subscription successful.
ERRORLOG/1 @ initPackageFromTopicList, L167: package [0] is being occupied.

STATUS/1 @ reset, L567: Reset Subscription Successful.`

Can you help me?(

wstefan20, 2020-10-19:
I have this same error and I'm running a full A3 with lightbridge. I hope they solve this.

dji-kyle, 2021-06-24:
Note: 
1、ensure the FC in P mode 
2、If using third-party RC，the RC joystick should be back to center.

closed 2021-06-24 by dji-kyle