dji-sdk/Onboard-SDK #446: AdvancedSensing : spam error but is working fine

depauwjimmy, 2019-06-26:
I believe it started with the 3.8.1 update but it could be 3.8 as well.
As soon as i start a camera stream using startMainCameraStream i received a lot of:
ERRORLOG/1 @ sendData, L189:
or sometimes :
ERRORLOG/1 @ sendData, L189: Port closed

The stream seems to be working fine so i ignore them for now.
What's strange is the error continues after i sent stopMainCameraStream and never stops until i completely shutdown my program.

Obviously i wanted to check the firmware since the readme reports another version

M210/M210 RTK | 1.2.0440 | 3.3.10.12

But this version does not seem to exist anywhere.
DJI Assistant does not list it and there is no reference of it whatsoever.
I could not find a page where the last firmware version of M210 is supposed to be.

Thanks

dji-dev, 2019-06-27:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/25446'>Zendesk ticket #25446</a>:<br><br>Dear Customer,<br><br>Thank you for contacting DJI.<br><br>In order to better understand and help you solve the problem, suggest you click the hyperlink on the right to fill in and submit more complete information about the problem.<br>https://formcrafts.com/a/dji-developer-feedback-en<br>The problem description column shall be as detailed as possible, including the following contents:<br>1. The type of camera you are using<br>2. You use onboard computer models and platforms<br>3. The complete log of program running or testing (can be added to the attachment)<br>4. Other screenshots or demo video and other necessary information (can be added to the attachment)<br><br>Thank you for your understanding and support, hope you a delighted life.<br><br>Best Regards,<br>DJI SDK Support Team

dji-dev, 2019-06-27:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/25446'>Zendesk ticket #25446</a>:<br><br>Dear Customer,<br><br>Thank you for contacting DJI.<br><br>As for the firmware problem of M210/RTK v1, suggest DJI Assistant 2 as the standard. sorry for the trouble caused to you.<br><br>Thank you for your understanding and support, hope you a delighted life.<br><br>Best Regards,<br>DJI SDK Support Team

closed 2021-06-24 by dji-kyle