dji-sdk/Onboard-SDK #447: Stack Corruption in Vehicle::parseDroneVersionInfo()

SChalice, 2019-07-16:
```
  uint8_t crc_id[16] = {};
  int     i          = 0;
  while (*ackPtr != '\0')
  {
    crc_id[i] = *ackPtr;
    i++;
    ackPtr++;
    if (ackPtr - startPtr > 18)
    {
      return false;
    }
  }
```

  uint8_t crc_id[16] = {};

needs to be

  uint8_t crc_id[17] = {};

SChalice, 2019-07-16:
dji-dev edited crap into my post for some reason, another bug....

dji-dev, 2019-07-17:
> Public comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/26094'>Zendesk ticket #26094</a>:<br><br>Thank you for your feedback. Our R&D engineers are looking into your questions now!. If there is any feedback I will contact you as soon as possible.

dji-dev, 2019-07-19:
> Public comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/26094'>Zendesk ticket #26094</a>:<br><br>Thank you for your feedback. We will fix it in later version. Please pay attention to the status of the issue.
In addition, about "edited somthing into your post" issue, it is a bug of the plugin and we have fixed it.

closed 2021-06-24 by dji-kyle