dji-sdk/Onboard-SDK #451: Onboard SDK and GSPro app coexistence.

hyonlim, 2019-07-18:
Hi,
We are going to implement obstacle avoidance while performing pre-defined mission by GS Pro.
How could this be implemented? Can we interrupt control authority while the vehicle is performing mission designed by GS Pro?

dji-dev, 2019-07-18:
> Public comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/26168'>Zendesk ticket #26168</a>:<br><br>Hi. Is your issue related to the onboard SDK? Or it is only related to the use of GS pro.

hyonlim, 2019-07-18:
Onboard SDK.

dji-dev, 2019-07-19:
> Public comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/26168'>Zendesk ticket #26168</a>:<br><br>Onboard SDK can interrupt control authority while the vehicle is performing mission designed by GS Pro. Just call vechile->obtainCtrlAuthority().

closed 2021-06-24 by dji-kyle