dji-sdk/Onboard-SDK #452: Matrice 100 keeps rotating when using "attitudeAndVertPosCtrl()"

ashibhuvan, 2019-07-18:
Hello,

Currently I am using Onboard-SDK for the Matrice 100. My onboard-computer is Raspberry Pi 3 and I am connected through serial. I have a problem that I want to use attitudeAndVertPostCtrl to change the roll and pitch of the vehicle. In simulation my code works perfectly but it always works in simulation. In real life I notice the drone keeps rotating regardless of what value I set for yaw. Does anyone know why this happens? Some help would greatly be appreciated. This is an example of the code I use.

`   vehicle->control->attitudeAndVertPosCtrl(0, 0, 0,
                                0);
              usleep(20000); `

I do not understand why this command would cause the vehicle to rotate. Any help is appreciated.

ashibhuvan, 2019-07-18:
And I have code running in a closed loop however it does not make a difference if I send this once or in a loop at 50Hz.

dji-dev, 2019-07-19:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26197'>Zendesk ticket #26197</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

To better understand and help you solve the problems, we need know more information below:
1. DJI M100 firmware version? recommended to update to the latest
2. DJI Onboard SDK version? recommended that using OSDK 3.8.1
3. Can you provide the program you tested? And a brief explanation
4. Recommended that provide the flight data of actual flight test (exported from DJI Assistant2)

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

ashibhuvan, 2019-07-19:
Hello,

I am using latest version of firmware availiable and sdk. The program is just for testing control ability but I put a sample below where if the input is "w", There is command send for .5 second. I get behaviour from the drone but does not make sense why it rotates and then rotates again in the opposite direction. If I have different pitch values but same yaw as I do in every command, there is different rotations. And in the simulations there are no rotation, in fact my code always works perfectly in simulation but in real life is incredibly different. I will try to do some more testing and export the flight data. I have also been taking videos to understand the drone movement. 

`if(input == 'w'){
        std::cout << "forward\n" << std::endl;
        roll = 0.0f;
        pitch = 5.0f;

       
      
      for (int i = 0; i < 25; i++)
           {
             //vehicle->control->velocityAndYawRateCtrl(-2.0,0.0,0.0,0.0);
              vehicle->control->attitudeAndVertPosCtrl(roll, pitch, yaw, elevation);
              usleep(20000); 
            
           }
           
          vehicle->control->attitudeAndVertPosCtrl(0,0,yaw,.5);`

dji-dev, 2019-07-19:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26197'>Zendesk ticket #26197</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

There is nothing wrong with the code, and you have shown that the simulator is working properly.
So it is recommended that you can submit flight data of the actual testing to better help you analyze and solve the problem.(exported from DJI Assistant2)

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

ashibhuvan, 2019-07-19:
Hello,

It is okay I have figured out. I think this has to do with the fact that the yaw angle is being refrenced to a different frame ( the world frame) but I am not too sure. This is how I fixed it. You can manually do this by modification of the control flag bits and then sending those commands but basically it is to change the type of yaw from YAW_ANGLE to YAW_RATE. I did this by modifying dji_control.cpp

```
void
Control::attitudeAndVertPosCtrl(float32_t roll, float32_t pitch, float32_t yaw,
                                float32_t z)
{
  //! @note 18 is the flag value of this mode
  uint8_t ctrl_flag =
    (VERTICAL_POSITION | HORIZONTAL_ANGLE | YAW_RATE | HORIZONTAL_BODY);
  CtrlData data(ctrl_flag, roll, pitch, z, yaw);

  this->flightCtrl(data);
```

YAW_RATE was originally YAW_ANGLE. Now you can pass in 0 as a command and no rotation will occur as you are sending in the rate modification command. If you do not want to modify the SDK just create a ctrl_flag like is created in the SDK above and send to flightCtrl your data. Let me know if anyone else needs help with this. Closing issue.

closed 2019-07-19 by ashibhuvan

dji-dev, 2019-07-22:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26197'>Zendesk ticket #26197</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Your solution is also correct, congratulations on solving the problem and good luck with the project.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team