dji-sdk/Onboard-SDK #453: DJI onboardSDK for Matrice 600 Pro

jackie0052, 2019-07-21:
Hi,

We are currently developing an iOS app using the mobileSDK to communicate with a Raspberry Pi onboard a Matrice 600 pro. The onboardSDK we are using now is Version 3.2 and our drone firmware version is 1.00.01.66. When I tried to connect the Raspberry Pi to the drone, it shows me the following error:

_ERROR parseDroneVersionInfo, line 271: Drone version was not obtained. Please restart the program or call getDroneVersion._

I am pretty sure that the UserConfig.txt file that I am using was set correctly. We have successfully connected the Raspberry Pi to the drone before without any problem. This error just showed up recently. 

Any suggestion on how to debug the problem will be highly appreciated.
[image]

Jai-GAY, 2019-07-22:
Not sure is your case similar to this. [Stack Corruption in Vehicle::parseDroneVersionInfo()](https://github.com/dji-sdk/Onboard-SDK/issues/447#issuecomment-512094584)

jackie0052, 2019-07-22:
> Not sure is your case similar to this. [Stack Corruption in Vehicle::parseDroneVersionInfo()](https://github.com/dji-sdk/Onboard-SDK/issues/447#issuecomment-512094584)

Thanks, Jai! I am not sure. 

Do you know where I can find the following information posted in issue #447: 

_uint8_t crc_id[16] = {}_

Thanks agian.

dji-kyle, 2021-06-24:
Update the OSDK version to 3.9 and firmware version to the latest.
The following forum community is recommanded for you, if the problem persists, or new questions about the DJI SDK,
https://djisdksupport.zendesk.com/hc/en-us/community/topics

closed 2021-06-24 by dji-kyle