dji-sdk/Onboard-SDK #454: osdkcompiled error

david6096, 2019-07-23:
I compiled the osdk with the following steps:
mkdir build 
cd build
cmake .. -DADVANCED_SENSING=ON -DWAYPT2_CORE=ON
make
Scanning dependencies of target advanced-sensing
[  1%] Creating directories for 'advanced-sensing'
[  2%] Performing download step (git clone) for 'advanced-sensing'
Cloning into 'advanced-sensing'...
Scanning dependencies of target advanced-sensing
[  1%] Creating directories for 'advanced-sensing'
[  2%] Performing download step (git clone) for 'advanced-sensing'
Cloning into 'advanced-sensing'...

and freezed in this step,can't cloning the source code .
some tips?

david6096, 2019-07-24:
the compiled platform is raspberry pi 3,when compiled with ADVANCED_SENSING=ON,the downloaded file is advanced-sensing-1.0.0.tar.gz,but the needed file is advanced-sensing-2.0.3.tar.gz.

david6096, 2019-07-24:
the log messages of make:
Scanning dependencies of target advanced-sensing
[  1%] Creating directories for 'advanced-sensing'
[  2%] Performing download step (git clone) for 'advanced-sensing'
Cloning into 'advanced-sensing'...
Switched to a new branch 'advanced-sensing-2.0.3-armv7'
Branch advanced-sensing-2.0.3-armv7 set up to track remote branch advanced-sensing-2.0.3-armv7 from origin.
[  3%] No patch step for 'advanced-sensing'
[  4%] No update step for 'advanced-sensing'
[  5%] Performing configure step for 'advanced-sensing'
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Onboard-SDK-3.8/build/osdk-core/AdvancedSensing
[  6%] Performing build step for 'advanced-sensing'
[  7%] No install step for 'advanced-sensing'
[  7%] Completed 'advanced-sensing'
CMake Error: Problem with archive_read_open_file(): Failed to open '/home/pi/Onboard-SDK-3.8/build/osdk-core/advanced-sensing-prefix/src/advanced-sensing/advanced-sensing-2.0.3.tar.gz'
CMake Error: Problem extracting tar: /home/pi/Onboard-SDK-3.8/build/osdk-core/advanced-sensing-prefix/src/advanced-sensing/advanced-sensing-2.0.3.tar.gz
osdk-core/CMakeFiles/advanced-sensing.dir/build.make:125: recipe for target 'advanced-sensing' failed
make[2]: *** [advanced-sensing] Error 1
CMakeFiles/Makefile2:124: recipe for target 'osdk-core/CMakeFiles/advanced-sensing.dir/all' failed
make[1]: *** [osdk-core/CMakeFiles/advanced-sensing.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

david6096, 2019-07-24:
https://github.com/dji-sdk/Onboard-SDK-Resources/tree/advanced-sensing-2.0.3-armv7,there is an old 1.0.0 version in this branch,it need to be rebuild.please to fix this bug.thanks.

dji-dev, 2019-07-26:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26328'>Zendesk ticket #26328</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

You may first check whether the terminal of your device can be directly accessed to github. It may take some time to download. And how long you have been waiting?
Also, if you cannot connect due to network reasons, it is recommended to download the “advanced-sensing” package directly to github link (https://github.com/dji-sdk/Onboard-SDK-Resources) and manually unzip it to the specified directory (shown below):
 [image]
DWAYPT2_CORE, can do the same thing

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle