dji-sdk/Onboard-SDK #455: Increase maximum subscription rate of TOPIC_GPS_FUSED

L4ncelot, 2019-07-25:
Hi,

I'm currently using older version 3.7 of Onboard-SDK in combination with Onboard-SDK-ROS. For my application I need to subscribe data from `TOPIC_GPS_FUSED` with rate of 100 Hz.  

According to the [dji_subscription.cpp](https://github.com/dji-sdk/Onboard-SDK/blob/3.8/osdk-core/api/src/dji_subscription.cpp#L55) the maximum rate can be only 50 Hz. 

So basically what I've done was to change this value naively to 100 Hz. Then I've added the `TOPIC_GPS_FUSED` subscription topic to the topicList100Hz [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/dji_sdk/src/modules/dji_sdk_node.cpp#L429) so that the rate of ROS topic will be 100 Hz.

Everything seems to be working but there is a reason why the maximum rate was set to 50 Hz. So my question therefore is if the firmware can actually handle this topic on 100 Hz or not. How can I even test this?

Thank you for any advice.

dji-dev, 2019-07-26:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26439'>Zendesk ticket #26439</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

`TOPIC_GPS_FUSED` 50HZ is determined by flight control. Even if you set the publish rate at 100hz, the actual update frequency is 50HZ at the highest.
As for the highest frequency of data of each TOPIC, please refer to : https://developer.dji.com/cn/onboard-api-reference/group__telem.html#ga563d2333dc43253129a3915970f99cfaa4970ab673d0bf016fab967785e3a5141  (it may be helpful to you)

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

L4ncelot, 2019-07-26:
@dji-dev thank you very much for clarifying that.

closed 2019-07-26 by L4ncelot

dji-dev, 2019-07-26:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26439'>Zendesk ticket #26439</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It's our pleasure to answer your questions, hope you a delighted life.

Best Regards,
DJI SDK Support Team