dji-sdk/Onboard-SDK #456: Flight control rate restriction to 50 Hz

L4ncelot, 2019-07-25:
Hi,

I'm currently using older version 3.7 of Onboard-SDK in combination with Onboard-SDK-ROS and A3 autopilot. For my application I need to send the control commands to the autopilot with the rate of 100 Hz.

For this I'm using the `flightCtrl` function declared [here](https://github.com/dji-sdk/Onboard-SDK/blob/3.8/osdk-core/api/inc/dji_control.hpp#L387). 

I've tested this only in the simulation but there is a problem with the behaviour of autopilot. It seems like the autopilot takes over the control randomly (or at least it ignores control commands) which messes with my application. After some time it releases the control (or stops ignoring it) but my application is already messed up so it doesn't matter.

After some testing and researching I've found some documentation [here](https://developer.dji.com/onboard-sdk/documentation/M100-Docs/application-development-guides/programming-guide.html). In the section **"Movement Control"** it says: "We recommend developers send their movement control commands at 50Hz frequency.". So I tried to ignore half of my inputs and everything started working.

Is this firmware limitation or is there anything I can do about it to make it work on 100 Hz?

Thank you for any advice.

Jai-GAY, 2019-07-26:
Have you try to look [into here?](https://github.com/dji-sdk/Mobile-SDK-Android/issues/421#issuecomment-510778630)

L4ncelot, 2019-07-26:
@Jai-GAY, thanks for response. But the issue is related to the subscription of incoming messages from autopilot. For the control input there's nothing mentioned about the rate other than the note I've posted earlier. So I wanted to make sure it's firmware issue and nothing can be done even if I modify the osdk-core.

dji-dev, 2019-07-26:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26442'>Zendesk ticket #26442</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

This is determined by flight control. The maximum loop rate of control functions can be referred to as follows, which may be helpful to you (100HZ control frequency can be achieved by using angularRate and attitude control methods).

    Control::positionAndYawCtrl        # The highest frequency 50Hz
    Control::velocityAndYawRateCtrl    # The highest frequency 50Hz
    Control::attitudeAndVertPosCtrl    # The highest frequency 200Hz  But the vertical position information is only updated at 50Hz
    Control::angularRateAndVertPosCtrl # The top cycle frequency is 400Hz but the vertical position information is only updated at 50Hz
    Control::flightCtrl                # According to the incoming "flag", the highest update frequency of angularRate information is 400Hz, attitude information is 200Hz, speed and position information is 50Hz.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

L4ncelot, 2019-07-26:
@dji-dev, thank you so much for the frequencies. But it's weird because I'm using this flag to control the copter:
```
uint8_t flag = (Control::HORIZONTAL_ANGULAR_RATE | Control::VERTICAL_THRUST | Control::YAW_RATE |
                    Control::HORIZONTAL_BODY | Control::STABLE_DISABLE);
```
I should have mention from the beginning I'm using angular rates and thrust as control commands. So basically it should be possible right? 

But why it works with 50 Hz and not 100 Hz? I have the autopilot connected to my PC through FTDI and I had some problems with delays of it's buffer previously. So I had to set the receiving buffer for low latency using `setserial /dev/ttyUSB0 low_latency`. Could it be perhaps the same problem but on the sending side causing the autopilot to behave as I described?

L4ncelot, 2019-07-26:
@dji-dev one more question. Is the information about the frequencies available somewhere? Because I cannot find it.

dji-dev, 2019-07-26:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26442'>Zendesk ticket #26442</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Can you provide the complete code of your tests? And the specific phenomena of confusion (such as program log, better with video)
or can you try to use "angularRateAndVertPosCtrl" control to do that?

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

L4ncelot, 2019-08-14:
@dji-dev sorry for the delay of my response. I was working on different stuff for while. I couldn't post my code because it's complex environment controlling the copter. 

But I've eventually solved the problem with "stutter" movement of the vehicle. It was really the limitation of serial link between PC and autopilot. After removing some messages we don't particularly need and modifying some frequencies it started working perfectly. 

The problem wasn't on your side. Thank you for helping me solve the problem.

closed 2019-08-14 by L4ncelot