dji-sdk/Onboard-SDK #457: recompile with -fPIC

roseyangyu, 2019-07-29:
[image]
How to solve that?

dji-dev, 2019-07-29:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26519'>Zendesk ticket #26519</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

In order to better understand and help you solve the problem, need to know more information as follows:
1. What is the OSDK version you use? recommended use OSDK-3.8.1
2. What platform are you compiling on? X86 - ubuntu?
3. Can you provide the complete log to the attachment?
4. Ensure that the versions of relevant tools meet the requirements, as shown in the following figure:
 [image]​

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/odQl78nTgczpoNMP2YOenbrU3' target='zd_attach'>inline2093400386.png</a>

roseyangyu, 2019-07-30:
just support ros-kinetic?I used ubuntu18.04 and ros-melodic.

dji-dev, 2019-07-30:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26519'>Zendesk ticket #26519</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We recommend using ubuntu16.04 + ROS kinetic. Other environments are untested and cannot guarantee its stability.
 [image]​

Best Regards,

DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/Bp7ha5VfxXj4Gv0nLSKZdyM0Z' target='zd_attach'>inline-1256958689.png</a>

roseyangyu, 2019-07-30:
my god！Opencv-3.3 just install by compile? I can't install by apt

roseyangyu, 2019-07-30:
I want to use Onboard-SDK-ROS,In Onboard-SDK-ROS readme,"The ROS package requires Onboard SDK (djiosdk-core) to be installed to your system prior to running it",Can I just install djiosdk-core?I don't compile sample?

roseyangyu, 2019-07-30:
我需要用Onboard-SDK-ROS，github上说需要安装OSDK，但是只是djiosdk-core，我是不是只要把djiosdk-core编译出来就可以了，不需要编译sample
 

------------------

杨禹

北京中飞艾维航空科技有限公司/Xlab

北京市大兴区凉水河二街8号大族企业湾16号楼B座二层

 
 
 
------------------ Original ------------------
From: "DJI"; 
Date: 2019年7月30日(星期二) 下午4:36
To: "dji-sdk/Onboard-SDK"; 
Cc: "Elite yang"; "Manual"; 
Subject: Re: [dji-sdk/Onboard-SDK] recompile with -fPIC (#457)

 
 
Public comment from Bruce Cheung in Zendesk ticket #26519:

Dear Customer,
  
Thank you for contacting DJI.
 
We recommend using ubuntu16.04 + ROS kinetic. Other environments are untested and cannot guarantee its stability.
 ​
 
Best Regards,
 
DJI SDK Support Team
inline-1256958689.png

dji-dev, 2019-07-31:
> Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26519'>Zendesk ticket #26519</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

Opencv 可以直接使用  apt-get install libopencv-dev
OSDK-ROS3.8.1 的编译步骤已经包含djiosdk-core的编译和安装，您可以参照OSDK的开发文档进行安装OSDK-ROS：
https://developer.dji.com/onboard-sdk/documentation/development-workflow/sample-setup.html#ros-onboard-computer
如果您只是编译和安装djiosdk-core，还是需要安装OSDK-ROS，不然意味着您需要自己实现ROS node

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

closed 2021-06-24 by dji-kyle