dji-sdk/Onboard-SDK #462: VirtualRC control not working.

ashibhuvan, 2019-08-06:
Hello,

Hope all is well. I am trying to do some testing with the VirtualRC protocol but it seems that when I try to use it, I see no activity from my drone. I have tried both this in the simulator and in real life. Below I have attached the code I run. During the compilation I get no errors but when I run the code I get no movement. 

```
    
            VirtualRCData bob;
            vehicle->virtualRC->setControl(true,VirtualRC::CutOff_ToRealRC);

            bob.roll = 1684;     //0-> roll      [1024-660,1024+660] 
            bob.pitch = 1684;    //1-> pitch     [1024-660,1024+660]
            bob.throttle = 1024+660;  //2-> throttle  [1024-660,1024+660]
            bob.yaw = 1024;       //3-> yaw       [1024-660,1024+660]
            bob.gear = 1324;      //4-> gear    {1684(UP), 1324(DOWN)}
            bob.mode = 1552;      //6-> mode      {1552(P), 1024(A), 496(F)}

           for(int i = 0; i < 60; i++){
                std::cout << "sending somedata" << std::endl;
                if(vehicle->virtualRC->isVirtualRC())
                {
                    vehicle->virtualRC->sendData(bob);
                    vehicle->virtualRC->getVRCData();
                }
                usleep(200000);
            }

```
The documentation says to use  uint32_t virtual_rc_data[16] but I have tried this and it does not work as it says that the sendData function for virtualRC does not accept a uint32 [16] as a parameter. In the class file for VirtualRC, the struct is declared as I have written below.

Any help with this would be greatly appreciated.

dji-dev, 2019-08-07:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26811'>Zendesk ticket #26811</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

To better understand and help you solve the problem, more information is needed below:
1. Are you using M100? Firmware version?
2. Are you based on OSDK or a protocol program created by yourself? Can you provide a complete test code file?

It is suggested that you can click the hyperlink on the right to fill in and submit more detailed problem information, including test code (uploaded to the attachment). https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

ashibhuvan, 2019-08-07:
yes, the drone is M100. I am using the OnboardSDK. The code snippet is literally all of the code basically. and of course i am using 1.3.1.82 , i dont think there has been a newer version for over a year now of the firmware. could you please check with the engineers maybe to see if there is anything wrong with this code? I will try at different frequencies as well.

dji-dev, 2019-08-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26811'>Zendesk ticket #26811</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The question needs further analysis and testing, if need further communication or there is any result, we will contact you immediately.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

ashibhuvan, 2019-08-09:
Thank you very much for all your support. I am spending many many hours working on coding with the OnboardSDK and it would be nice to know if this is actually supported or not. A ton of the documentation is not updated at all and it is very frustrating to spend so much time when DJI says the code is supposed to work one way but it works entirely another way.

dji-dev, 2019-08-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26811'>Zendesk ticket #26811</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry for the inconvenience caused to you. We will improve the document in the future. In addition, regarding flight control of M100, could you consider using flight-control API? https://developer.dji.com/onboard-api-reference/classDJI_1_1OSDK_1_1Control.html#a1f2c5a277f091fb5fb5b39baf6f6fa0f

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

ashibhuvan, 2019-08-12:
Yes if that is confirmation that the VirtualRC protocol does not work with the latest firmware that is perfectly fine. I would rather have a clear response later on for the status of VirtualRC because I think it would be important to know the exact capabilities of the system we have purchased to be more efficient. Thank you for all your support, your team is always very helpful and quick with the responses.

dji-dev, 2019-08-13:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26811'>Zendesk ticket #26811</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Understand your requirement, we will give you a definite result as soon as possible.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

ashibhuvan, 2019-08-19:
Hello,

Hope all is well. Any update on VirtualRC? 

Thanking You

dji-dev, 2019-09-03:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/26811'>Zendesk ticket #26811</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry to have kept you waiting, after further verification and confirmation, the firmware of DJI M100 can no longer support the Virtual RC protocol. It is suggested that you can use the FlightCtrl API to achieve control.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle