dji-sdk/Onboard-SDK #465: Sensor fusion of TOPIC_GPS_FUSED

surgeonofdeath, 2019-08-27:
I am using an DJI A3 Pro which has 3 GPS sensors and 3 IMU sensors for triple redundancy. I am wondering whether the sensor values received on TOPIC_GPS_FUSED is the data fused for all 6 sensors or only 1 GPS and 1 IMU, and the others are just there for redundancy? 

Thank you in advance.

surgeonofdeath, 2019-08-30:
Thank you for your reply. 

I am currently using the A3 Pro flight controller for a task in which RTK and the DGPS (Differential GPS) algorithm is prohibited. I am wondering if you can tell me if the A3 uses DGPS algorithm? I do not have the D-RTK device. I only have 3 GPS and 3 IMU. If you could just clarify this one doubt, I will bee clear whether to use the DJI A3 or not.

Thank you for your time.

dji-dev, 2019-08-30:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27516'>Zendesk ticket #27516</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that after further confirmation, when A3 connects multiple GPS modules, the purpose is to be redundant. The currently used GPS module will have a green light to indicate, without using the algorithm you mentioned.

Thank you for your understanding and support, hope you a delighted life.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle