dji-sdk/Onboard-SDK #466: Make logging output redirectable

hdino, 2019-09-03:
For several applications it might be very useful to be able to redirect the logging output. I would suggest to implement something like:
```C++
void __attribute__((weak)) printDjiLog(const char * format, va_list arg)
{
    vprintf(format, arg);
}

Log&
Log::print(const char* fmt, ...)
{
  if ((!release) && vaild)
  {
    va_list args;
    va_start(args, fmt);
    mutex->lock();
    printDjiLog(fmt, args);
    mutex->unlock();
    va_end(args);
  }
  return *this;
}
```
in [dji_log.cpp](https://github.com/dji-sdk/Onboard-SDK/blob/908f91a7ff27655a558ad2c69b87d8ffd42631de/osdk-core/hal/src/dji_log.cpp#L108), such that the user can easily reimplement the `printDjiLog` function.

dji-dev, 2019-09-04:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27707'>Zendesk ticket #27707</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Your good advice was very much appreciated.
Also, we would like to know some USES for which you need to use log redirection?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

hdino, 2019-09-04:
Well, I guess most users of the Onboard-SDK have an embedded system on the aircraft and a normal PC on the ground. Then it would be nice to see the logging output of the Onboard-SDK on the groundstation by including those log messages in the data link between aircraft and groundstation, which could be based on many technologies.

dji-dev, 2019-09-04:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27707'>Zendesk ticket #27707</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Thank you for your feedback and we will make further analysis and evaluation of your advice. We will inform you as soon as there is any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2019-09-06:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/27707'>Zendesk ticket #27707</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Thank you for your feedback and patience. As for your requirements, we will consider adding them into the later support plan. However, as you know, software development needs certain time and uncertainty, so we cannot guarantee the expected release version and time with you. It is recommended that you continue to follow the release and product trends of OSDK.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle