dji-sdk/Onboard-SDK #469: Run ./djiosdk-flightcontrol-sample UserConfig.txt error

liu978079474, 2019-10-08:
I install Onboard_sdk successfully,but I run  ./djiosdk-flightcontrol-sample UserConfig.txt error.
log:
Read App ID
User Configuration read successfully. 

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

ERRORLOG/1 @ _serialOpen, L175: cannot open device /dev/ttyUSB0

ERRORLOG/1 @ init, L60: ...Failed to start serial
Comms appear to be incorrectly set up. Exiting.
Segmentation fault (core dumped)

dji-dev, 2019-10-08:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28627'>Zendesk ticket #28627</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

What development platform are you using?
The error info indicates that the serial device cannot be opened. Are you using the usb-TTL device?
And you can refer to the DJI OSDK development documentation for the Hardware Setup Guide section https://developer.dji.com/onboard-sdk/documentation/development-workflow/hardware-setup.html

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

liu978079474, 2019-10-08:
@dji-dev Hi,I used the NVIDIA TX2 Ubuntu16.04 development platform, and I run ./M210ConfigTool --usb-port /dev/ttyACM0 --config-file UserConfig.txt --simulation on --latitude 37.422083 --longitude -122.137390.
Log show:bash: ./M210ConfigTool: cannot execute binary file: Exec format error

liu978079474, 2019-10-08:
@dji-dev Hi,I could not find /dev/ttyUSB0 in the NVIDIA TX2.

dji-dev, 2019-10-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28627'>Zendesk ticket #28627</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

1. I used the NVIDIA TX2 Ubuntu16.04 development platform, and I run ./M210ConfigTool --usb-port /dev/ttyACM0 --config-file UserConfig.txt --simulation on --latitude 37.422083 --longitude -122.137390. Log show:bash: ./M210ConfigTool: cannot execute binary file: Exec format error
==> You need to use the executable file under the arm v8 architecture and ensure execution permissions.
https://github.com/dji-sdk/Onboard-SDK/tree/master/utility/bin/armv8/64-bit

2. I could not find /dev/ttyUSB0 in the NVIDIA TX2.
==> If you want to use /dev/ttyusb0 and you need to connect to a USB-TTL device, make sure you are connected to the device.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle