dji-sdk/Onboard-SDK #470: Vehicle class cannot be destructed if no data is received from the serial port

hdino, 2019-10-08:
Hi,
I found a bug in the Onboard-SDK (current version, 3.9) for Linux:

If the flight controller is NOT connected to the serial port, the activation of the vehicle will fail, but the vehicle class cannot be destructed.

Minimal example:
```c++
bool dji_thread_support = true;
auto dji_vehicle_ = std::make_shared<DJI::OSDK::Vehicle>(
    serial_port.c_str(), baudrate, dji_thread_support);

if (!dji_vehicle_->protocolLayer->getDriver()->getDeviceStatus())
        THROW_EXCEPTION("DJI communication error");

DJI::OSDK::Vehicle::ActivateData activate_data;
// activate_data is filled here

auto dji_ack = dji_vehicle_->activate(&activate_data, dji_config_->timeout);

if (DJI::OSDK::ACK::getError(dji_ack)) {
    DJI::OSDK::ACK::getErrorCodeMessage(dji_ack, __func__);
    dji_vehicle_.reset(); // ERROR OCCURS HERE
}
```

Reason:
The destructor of the vehicle class tries to stop the serial port read thread in [dji_vehicle.cpp:364](https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/osdk-core/api/src/dji_vehicle.cpp#L364). Then the `stopThread` command tries to join the thread, see [posix_thread.cpp:117](https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/osdk-core/platform/linux/src/posix_thread.cpp#L117). However, the read command for the serial port blocks in [linux_serial_device.cpp:365](https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/osdk-core/platform/linux/src/linux_serial_device.cpp#L365) and thus the thread cannot be joined.

During the configuration of the serial port, both an intercharacter timeout of 0.1 seconds (`VTIME`) and a minimum number of 18 characters that have to be received (`VMIN`) are set in [linux_serial_device.cpp:301-302](https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/osdk-core/platform/linux/src/linux_serial_device.cpp#L301). However, the intercharacter timer will never kick in if no characters are received and thus the read command will block infinitely.

dji-dev, 2019-10-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28648'>Zendesk ticket #28648</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Thank you for your feedback and suggestions. We will make further analysis and verification of the problems you raised and will inform you as soon as there is any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2019-11-05:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28648'>Zendesk ticket #28648</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The problem has been confirmed and is expected to be fixed in the next version. Thank you for your feedback.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle