dji-sdk/Onboard-SDK #472: 请问这个错误是什么意思: ERRORLOG/1 @ getValue, L315: Topic 0xA value memory not initialized, return default

addriumruss, 2019-10-21:
这几个错误不停的出现, 不知道怎么回事:
ERRORLOG/1 @ getValue, L315: Topic 0xA value memory not initialized, return default
ERRORLOG/1 @ getValue, L315: Topic 0x1E value memory not initialized, return default
ERRORLOG/1 @ getValue, L315: Topic 0x1B value memory not initialized, return default
ERRORLOG/1 @ getValue, L315: Topic 0x19 value memory not initialized, return default
ERRORLOG/1 @ getValue, L315: Topic 0x18 value memory not initialized, return default

dji-dev, 2019-10-21:
Public comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/29025'>Zendesk ticket #29025</a>:<br><br>这几个错误不停的出现, 不知道怎么回事:
ERRORLOG/1 @ getValue, L315: Topic 0xA value memory not initialized, return default
ERRORLOG/1 @ getValue, L315: Topic 0x1E value memory not initialized, return default
ERRORLOG/1 @ getValue, L315: Topic 0x1B value memory not initialized, return default
ERRORLOG/1 @ getValue, L315: Topic 0x19 value memory not initialized, return default
ERRORLOG/1 @ getValue, L315: Topic 0x18 value memory not initialized, return default

addriumruss, 2019-10-21:
@dji-dev 您好，这个页面 Zendesk ticket #29025 显示不存在啊
"
哎呀
您查找的页面不存在
您可能输错了网址，或者页面已被移动
"

dji-dev, 2019-10-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29025'>Zendesk ticket #29025</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

为了更好的了解和帮助您解决问题，建议您点击右侧的超链接填写和提交更完整的问题信息， https://formcrafts.com/a/dji-developer-feedback-cn
其中问题描述栏尽量详细，包括如下内容：
1. 使用的SDK版本，比如OSDK-3.9
2. SDK开发平台
3. 使用的DJI 产品型号
4. 运行的是原生sample吗？
5. 程序运行的完整log
6. 出现该问题的概率？100%还是其他？
7. 其它截图或者演示视频等必要信息

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

dji-kyle, 2021-06-24:
Note: 订阅项未初始化，
1、若有初始化可以在获取值前面增加点延时试试，若有改善问题可能与串口通信有关，可以尝试增大波特率。
2、注释报错点的TOPIC或package，检查是否有订阅机型不支持的TOPIC。

closed 2021-06-24 by dji-kyle