dji-sdk/Onboard-SDK #473: 启动发动机失败，请求支援啊!!

addriumruss, 2019-10-21:
每次启动发动机时总是失败，调用代码：
ACK::ErrorCode retStatus = vehicle->control->armMotors(linuxSetup.responseTimeout);
if (ACK::getError(retStatus)) {
	ACK::getErrorCodeMessage(retStatus, __func__);
}
错误:
STATUS/1 @ getErrorCodeMessage, L644: droneTakeoff
STATUS/1 @ getCommonErrorCodeMessage, L714: UNKNOWN_ACK_ERROR_CODE

用的最新板载SDK，请问怎么解决啊

dji-dev, 2019-10-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29030'>Zendesk ticket #29030</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

为了更好的了解和帮助您解决问题，建议您点击右侧的超链接填写和提交更完整的问题信息， https://formcrafts.com/a/dji-developer-feedback-cn
其中问题描述栏尽量详细，包括如下内容：
1. SDK开发平台
2. 使用的DJI 产品型号
3. 运行的是原生sample吗？
4. 程序运行的完整log
5. 出现该问题的概率？100%还是其他？
6. 如果要尝试复现您的问题，建议简要提供具体的操作步骤
7. 其它截图或者演示视频等必要信息

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

dji-kyle, 2021-06-24:
Note: 
1、检查飞机是否处于P模式
2、连接遥控器和APP，查看是否有报错信息
3、遥控器控制正常情况下，可以先断开遥控器连接看是否可以正常起飞。

closed 2021-06-24 by dji-kyle