dji-sdk/Onboard-SDK #475: Custom action waypoint

retsef, 2019-10-29:
Hi,
it is possible create a custom action for each waypoint reached?

dji-dev, 2019-10-30:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29333'>Zendesk ticket #29333</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Yes, you can use CommandList and CommandParameter to implement custom actions, CommandList represents the list of waypoint actions. CommandParameter represents the list of parameters for a waypoint. For example, the code that needs to hover for 5 seconds after reaching the waypoint:
​ [image]​
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/1mWWx1FCsiXgHSjdIMguPK939' target='zd_attach'>inline1032639677.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/BhBfIvRuJUeUafdgt8KJsZgoo' target='zd_attach'>inline1304829195.png</a>

retsef, 2019-10-30:
Thanks @dji-dev for your answer, but I have to call
a custom function, **printMessage()**, as soon as the waypoint is reached. It's possible?

dji-dev, 2019-10-31:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29333'>Zendesk ticket #29333</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

You can call setWaypointEventCallback to register the callback function to determine if the aircraft has reached the waypoint, and call your custom function in the callback function.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

retsef, 2019-10-31:
Thanks @dji-dev.

closed 2019-11-02 by retsef

dji-dev, 2019-11-04:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29333'>Zendesk ticket #29333</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry, DJI Onboard SDK mainly supports C++, you can refer to the following demo (C++ implementation)
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/4Udkb4m5FnNOK2QZPnF22zYZ6' target='zd_attach'>inline-497165227.png</a>