dji-sdk/Onboard-SDK #476: 如何检测WayPointMission是否完成进行？

alankent911, 2019-10-30:
在DJI OSDK waypoint的simple里面只包含了创建并上传Waypoint mission list, 但我想知道是否可以检测任务状态， 比如说完成waypoint后返回一个值或是读取无人机飞行状态之类的， 十分感谢

dji-dev, 2019-10-30:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29348'>Zendesk ticket #29348</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

您可以调用setWaypointEventCallback来注册回调函数，以确定飞行器到达的航点位置和状态，代码示例如下：
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/h1UxQkO5GUVIPeKwypxq5qK3r' target='zd_attach'>inline-2138933357.png</a>

maxiao99, 2019-12-25:
按照上图方式用setWaypointEventCallback来注册回调函数，为什么回调函数不执行，是否还需要其他配置？
[image]

dji-kyle, 2021-06-24:
Note: 若注册回调代码正常调用，回调未触发，可以检查并勾选DJI Assistant2上OSDK界面的地面站状态推送。
[image]

closed 2021-06-24 by dji-kyle