dji-sdk/Onboard-SDK #477: inconsistent null return in OSDK v3.9.0

Jai-GAY, 2019-10-30:
Why this void* don't return null compare to the rest?

void* PosixThread::send_call(void* param)

[return NULL;](https://github.com/dji-sdk/Onboard-SDK/blob/295915e6da116aa2871760d3478bba7ebab2f5d9/osdk-core/platform/linux/src/posix_thread.cpp#L128)

dji-dev, 2019-10-30:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29353'>Zendesk ticket #29353</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

We have received the information you provided. We will make further analysis and verification of the problems you raised and will inform you as soon as there is any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-07-21:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29353'>Zendesk ticket #29353</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry to keep you waiting, because this function is a perpetual loop, and the condition of the loop is always true. It polls and sends all the time.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-06-24 by dji-kyle