dji-sdk/Onboard-SDK #482: M210 V2RTK连接图传USB，遥控器画面变灰，AdvancedSensing无法调用X5s相机流，显示Unable to read from MAIN_CAMERA lost, retry connecting。

Hust-hsg, 2019-11-16:
我使用妙算2进行OSDK开发，之前是一切正常的，在我运行改写的一段改写的camera-stream-callback-sample发生错误后，发生了如标题所说的情况。拔掉USB遥控器一切正常，但是插上后遥控器主相机画面变灰且静止，并提示sd卡空间为0。尝试过刷新固件，使用configtool设定可使用USB飞行等方法，但并未解决。

Hust-hsg, 2019-11-16:
当我运行例程时在指令尾部加入 /dev/ttyACM0 时，发现运行一段时间后能够调用图像，并且运行程序期间遥控器能显示主相机图像。但程序一旦运行结束遥控器主相机图像就静止变灰，且提示sd卡空间不足。并且Ubuntu界面提示Unable to mount 64GB Volume。

Hust-hsg, 2019-11-16:
./camera-stream-poll-sample UserConfig.txt /dev/ttyACM0
Please enter the type of camera stream you want to view
m: Main Camera
f: FPV  Camera
b: both Cameras
m
argv[1] = UserConfig.txt
Find the target ttyACM device , path:/dev/ttyACM0
Read App ID
User Configuration read successfully. 

Set ACM device:/dev/ttyACM0

STATUS/1 @ init, L56: Attempting to open device /dev/ttyTHS2 with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L762: Device Serial No. = 1DADG6400100S6

STATUS/1 @ parseDroneVersionInfo, L764: Hardware = PM420

STATUS/1 @ parseDroneVersionInfo, L765: Firmware = 3.4.3.37

ERRORLOG/1 @ initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ init, L49: Looking for USB device...

STATUS/1 @ init, L65: Found 7 USB devices, identifying DJI device...

STATUS/1 @ init, L83: Found a DJI device...

STATUS/1 @ init, L96: Attempting to open DJI USB device...

STATUS/1 @ init, L134: ...DJI USB device started successfully.

STATUS/1 @ activate, L1398: version 0x3040325

STATUS/1 @ activate, L1411: Activation successful

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ initGimbal, L926: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ init, L56: Attempting to open device /dev/ttyACM0 with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 Connect to MAIN_CAMERA successful

STATUS/1 All components for decoding initialized ...

STATUS/1 **** MAIN_CAMERA data reading thread start! ****

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

STATUS/1 Connect to MAIN_CAMERA successful

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

STATUS/1 Connect to MAIN_CAMERA successful
[... 274 lines cut ...]
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM
#### Got image from:	MAIN_CAM

Hust-hsg, 2019-11-16:
请问这是什么原因，希望能尽快解答。谢谢！！实验室催项目。

dji-dev, 2019-11-18:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29917'>Zendesk ticket #29917</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

[Q] 当我运行例程时在指令尾部加入 /dev/ttyACM0 时，发现运行一段时间后能够调用图像，并且运行程序期间遥控器能显示主相机图像。但程序一旦运行结束遥控器主相机图像就静止变灰，且提示sd卡空间不足。并且Ubuntu界面提示Unable to mount 64GB Volume。
==> 该情况属于正常现象，当接入USB时可以调用“./camera-stream-poll-sample”等sample获取相机视频流，此时遥控器图传也会恢复正常。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

Hust-hsg, 2019-11-18:
> Public comment from Bruce Cheung in [Zendesk ticket #29917](https://djisdksupport.zendesk.com/agent/tickets/29917):
> 
> 尊敬的客户，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> [Q] 当我运行例程时在指令尾部加入 /dev/ttyACM0 时，发现运行一段时间后能够调用图像，并且运行程序期间遥控器能显示主相机图像。但程序一旦运行结束遥控器主相机图像就静止变灰，且提示sd卡空间不足。并且Ubuntu界面提示Unable to mount 64GB Volume。
> ==> 该情况属于正常现象，当接入USB时可以调用“./camera-stream-poll-sample”等sample获取相机视频流，此时遥控器图传也会恢复正常。
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持

可是我之前不加入/dev/ttyACM0 ，能够调用图像流，而且无论连接还是不连接USB遥控器图传也一直正常啊。现在连接着USB遥控器就直接不能用了。。

Hust-hsg, 2019-11-18:
可之前连接着usb，遥控器是可以收到主相机的画面的啊。。

Hust-hsg, 2019-11-18:
> Public comment from Bruce Cheung in [Zendesk ticket #29917](https://djisdksupport.zendesk.com/agent/tickets/29917):
>
> 尊敬的客户，
>
> 您好！感谢您联系DJI大疆创新。
>
> [Q] 当我运行例程时在指令尾部加入 /dev/ttyACM0 [...]

那为什么我之前不加/dev/ttyACM0也能调用图像流，而且接着USB的时候，即使不运行例程遥控器也能显示X5s的图像呢？

dji-dev, 2019-11-19:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29917'>Zendesk ticket #29917</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

请问您之前使用的云台相机型号是什么？当使用X5S,X4S,X7时接入USB后是会出现该现象。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

Hust-hsg, 2019-11-19:
谢谢你的答复，请问只有这三款云台会出现这种情况吗？我看使用说明中说X5s需要接SD卡，是否是这几款云台的原因？我这里还有Z30，也是会出现这样的现象吗？

发送自 Windows 10 版邮件应用

发件人: DJI
发送时间: 2019年11月19日 19:45
收件人: dji-sdk/Onboard-SDK
抄送: Hust-hsg; Author
主题: Re: [dji-sdk/Onboard-SDK] M210 V2RTK连接图传USB，遥控器画面变灰，AdvancedSensing无法调用X5s相机流，显示Unable to read from MAIN_CAMERA lost, retry connecting。 (#482)

Public comment from Bruce Cheung in Zendesk ticket #29917:

尊敬的客户，
您好！感谢您联系DJI大疆创新。
请问您之前使用的云台相机型号是什么？当使用X5S,X4S,X7时接入USB后是会出现该现象。
感谢您的理解与支持,祝您生活愉快!
Best Regards，
DJI大疆创新SDK技术支持

snakehaihai, 2019-11-21:
Same thing happen to me. I got similar error as well

`STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat

STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat
[h264 @ 0x257f0a0] non-existing PPS 63 referenced
[h264 @ 0x2571400] non-existing PPS 63 referenced
[h264 @ 0x2571400] decode_slice_header error
[h264 @ 0x2571400] non-existing PPS 63 referenced
[h264 @ 0x2571400] decode_slice_header error
[h264 @ 0x2571400] non-existing PPS 63 referenced
[h264 @ 0x2571400] decode_slice_header error
[h264 @ 0x2571400] non-existing PPS 63 referenced
[h264 @ 0x2571400] decode_slice_header error
[h264 @ 0x2571400] no frame!
`

Hust-hsg, 2019-11-22:
Did you use m210?  Try to add “/dev/ttyACM0” in the end of “./***sample   UserConfig。TXT “。Dji said it is normal for RC showing no camera stream when  usb is pluged.

发送自 Windows 10 版邮件应用

发件人: snakehaihai
发送时间: 2019年11月21日 17:20
收件人: dji-sdk/Onboard-SDK
抄送: Hust-hsg; Author
主题: Re: [dji-sdk/Onboard-SDK] M210 V2RTK连接图传USB，遥控器画面变灰，AdvancedSensing无法调用X5s相机流，显示Unable to read from MAIN_CAMERA lost, retry connecting。 (#482)

Same thing happen to me. I got similar error as well
`STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat
STATUS/1 @ keep_camera_x5s_state, L44: Sending heart beat
[h264 @ 0x257f0a0] non-existing PPS 63 referenced
[h264 @ 0x2571400] non-existing PPS 63 referenced
[h264 @ 0x2571400] decode_slice_header error
[h264 @ 0x2571400] non-existing PPS 63 referenced
[h264 @ 0x2571400] decode_slice_header error
[h264 @ 0x2571400] non-existing PPS 63 referenced
[h264 @ 0x2571400] decode_slice_header error
[h264 @ 0x2571400] non-existing PPS 63 referenced
[h264 @ 0x2571400] decode_slice_header error
[h264 @ 0x2571400] no frame!
`

snakehaihai, 2019-11-23:
tried with and without ttyACM0   
Well that might have issue if gound operator can not verify what on board PC is seeing

Hust-hsg, 2019-11-23:
You meaned that your on board PC can recive camera stream  while RC can’t ?   
My problem is that both on board pc and RC can’t recive. I used ttyACM0 and  it worked out , but when camera-stream sample stop, it go to the same situation.
发送自 Windows 10 版邮件应用

发件人: snakehaihai
发送时间: 2019年11月23日 21:50
收件人: dji-sdk/Onboard-SDK
抄送: Hust-hsg; Author
主题: Re: [dji-sdk/Onboard-SDK] M210 V2RTK连接图传USB，遥控器画面变灰，AdvancedSensing无法调用X5s相机流，显示Unable to read from MAIN_CAMERA lost, retry connecting。 (#482)

tried with and without ttyACM0
Well that might have issue if gound operator can not verify what on board PC is seeing

dji-dev, 2019-11-26:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29917'>Zendesk ticket #29917</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

@snakehaihai, Have you solved the problem with the Camera stream? If not, suggest you click the hyperlink on the right to fill in and submit more complete information about the problem. https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

snakehaihai, 2019-11-27:
solve.
set a static IP 42.3/24/42.2 solved issue

MariusBeul, 2019-12-06:
I also get the

STATUS/1 Connect to MAIN_CAMERA successful
STATUS/1 Unable to read from MAIN_CAMERA lost, retry connecting ...

loop forever with an M210 v2 with Zenmuse X7 camera. Setting a static IP did not solve the issue. The 192.168.42.3 IP is correctly assigned.

liu978079474, 2020-05-13:
Hello, I also encountered the same problem. Have you have solve this problem？

MariusBeul, 2020-05-13:
Unfortunetely **NO**

snakehaihai, 2020-06-09:
try with osdk 4.01 then :)

closed 2021-06-24 by dji-kyle